iotsound/iotsound #265: DAC support: Adafruit I2S Audio Bonnet

tmigone, 2020-08-10:
This should do the trick, though we need confirmation it works:

- https://sound.balenalabs.io/docs/dac-configuration
- `BALENA_HOST_CONFIG_dtoverlay="hifiberry-dac”, “i2s-mmap”`

kevincastejon, 2020-09-10:
Just tried and bluetooth-audio is disabled at startup, I turn it on manually (by clicking the play button), I got this error everywhere spawning regularily 
`
snapcast-client  [Error] (worker) Exception in Controller::worker(): connect: Connection refused
`
even if I set DISABLE_MULTI_ROOM variable.
When I connect a device (my phone) and try to play audio i got these errors
`
10.09.20 03:46:52 (+0200)  bluetooth-audio  ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
10.09.20 03:46:52 (+0200)  bluetooth-audio  ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
10.09.20 03:46:52 (+0200)  bluetooth-audio  ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
10.09.20 03:46:52 (+0200)  bluetooth-audio  ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
10.09.20 03:46:52 (+0200)  bluetooth-audio  ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
10.09.20 03:46:52 (+0200)  bluetooth-audio  ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
10.09.20 03:46:52 (+0200)  bluetooth-audio  ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
10.09.20 03:46:52 (+0200)  bluetooth-audio  ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
10.09.20 03:46:52 (+0200)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such file or directory
10.09.20 03:46:52 (+0200)  snapcast-client  2020-09-10 01-46-52 [Error] (worker) Exception in Controller::worker(): connect: Connection refused
`
and no sound

tmigone, 2020-09-21:
hey Kevin, can you try again with the latest version of balenaSound (any greater than v3.0)? Thanks

kevincastejon, 2020-09-27:
I've achieved my project without balena since that post...
Luckily I still have a adafruit bonnet available but unfortunately no RPI0W, i'll try to find one to test that

tmigone, 2020-10-28:
Closing due to inactivity, feel free to reopen if the issue persists with a later version of balenaSound (> v3.0)

closed 2020-10-28 by tmigone