iotsound/iotsound #266: Multiple device aliases

OneBobone, 2020-08-17:
Hello,
Is it possible to have several active aliases for the DEVICE_NAME?
If the alias call can be identified, it would enable an interesting integration with multi-room amplifiers.

tmigone, 2020-08-17:
Hi there, can you expand a bit on this idea? Not sure I fully understand it.

OneBobone, 2020-08-17:
A multi zone amplifier typically takes a sources and amplifies the music to a selected pair of speakers in a zone. Some of these amplifiers can be controlled by code.

If I could publish on the network aliases of the DEVICE_NAME for each zone and track that selection, I could programmatically control the correct zone from a phone. Would be very nice.
Some systems do this already.