iotsound/iotsound #272: External DAC: Schiit Modi 3

Ben-Val, 2020-09-01:
I am using Raspberry Pi 4B with external Dac (Schiit Modi 3) and I see the balena sound connected on the Spotify App on my mobile phone, but no audio is coming out of the speakers even as I see the song is playing. Your assistance is much appreciated.

Ben-Val, 2020-09-01:
I'm just suspecting the external DAC is the reason why there is no sound output. Thanks

tmigone, 2020-09-03:
Hi @Ben-Val, is this a USB DAC? I mean what cable/connector is used to connect the DAC to the Pi?

Ben-Val, 2020-09-10:
Yes, the input to RPi is via USB. Thanks for the reply!

tmigone, 2020-09-21:
Hey @Ben-Val we recently released a new major version of balenaSound v3.0 which ships with out of the box support for USB soundcards. Give it a try, and let us know if it doesn't work. Cheers!

closed 2020-09-21 by tmigone