iotsound/iotsound #273: Audio Digione Map

zeltak, 2020-09-02:
Hi

was wondering if the project supported the Audio Digione Map?

https://github.com/balenalabs/balena-sound/issues/new

thx!

Z

tmigone, 2020-09-03:
Hi @zeltak, I couldn't find that device with a quick goole. Can you share a link?

zeltak, 2020-09-04:
Hi

sorry about that. Pasted this website instead of the actual project:

https://diyaudiocart.com/Allo-Digione

best

Z

tmigone, 2020-09-21:
Hi @zeltak, thanks for the link. We haven't tried that particular DAC though I see no reason for it not to work. If you are up to testing, it should be fairly easy to do so.

Here is a quick read on how to enable DAC's: https://sound.balenalabs.io/docs/audio-interfaces#dac-boards
You'll notice that the Allo Digione is not in that list (again because we haven't tested), but I found the `BALENA_HOST_CONFIG_dtoverlay` should be `allo-digione`. Let us know if this works and we can add it to the list. Cheers!

closed 2020-09-22 by tmigone