iotsound/iotsound #275: spotify: Consider removing Spotify Login

tmigone, 2020-09-03:
This came up on a forums thread: https://forums.balena.io/t/181405

Spotify login allows you to stream audio over the internet, so yo can stream from a device that's not connected to the local network. I'm not convinced that's a good enough reason to keep this, or is it? Unless we find a specific use case that makes sense we should look into removing the login option for Spotify. I tried to find a use case where this makes sense but nothing came to mind. 

@AlexProgrammerDE @chrisys let me know if you guys think otherwise or have an interesting use case. I'm not 100% sold on either.

AlexProgrammerDE, 2020-09-03:
I got one. Imagine your driving in a car and left the speaker at home active and you see it on your phone that its playing. You would basicly stop it remotely? Or if someone plays music over a second phone you should be able to stop it remotely? But yes it causes much issues and noone would really use it. But its a cool thing. @tmigone

tmigone, 2020-09-03:
@AlexProgrammerDE yeah that's a good one.

Another one from the forums:
```
I found out that for one guy finds it useful because of the multiroom or smth and he has speakers far back of his yard where he has a different wifi hotspot. So connecting via online is the best way for him
```

So it looks like it's a niche feature but still useful, thanks for the discussion, looks like we should keep it!

closed 2020-09-03 by tmigone