iotsound/iotsound #279: balena-sound support for the Hifiberry DAC2 HD

ehowell46, 2020-09-09:
The Hifiberry DAC2 HD was recently released, and is not on the balena sound configuration compatibility list. Would it be possible to add support for this device?

https://sound.balenalabs.io/docs/dac-configuration/

 https://www.hifiberry.com/shop/boards/hifiberry-dac2-hd/

Thank you!

Erik

doronbehar, 2020-09-12:
The balena sound project supports any dac that (usually Raspberry Pis) support. You will probably get along with setting the correct device tree overlay variable - in your fleet / device configuration. I got my Piano DAC working without surprise although it was not listed on that list.

Perhaps the issue isn't the lack of support, but the lack of documentation :).

ehowell46, 2020-09-12:
I've tried using the "hifiberry-dac" and "hifiberry-dacplus" overlays, but was unable to get them to work. I'll continue to try the remaining overlays available on the list.

doronbehar, 2020-09-12:
You should use `hifiberry-dacplushd` according to this: https://www.hifiberry.com/docs/data-sheets/datasheet-dac-hd/ .

ehowell46, 2020-09-12:
That worked! Thank you for your help with this! Very much appreciated. I will make sure to reference the product datasheet on Hifiberry's product page for dtoverlay information in the future.

tmigone, 2020-09-21:
Thanks @doronbehar and @ehowell46 for the testing. I'll add the DAC to the list. Cheers!

closed 2020-09-21 by tmigone