iotsound/iotsound #285: DEVICE_NAME and DISABLE_MULTI_ROOM environment variables seem to have no effect

rpecka, 2020-09-16:
**Describe the problem**
When I set the `DEVICE_NAME` and `DISABLE_MULTI_ROOM` environment variables as described in the main doc and the troubleshooting docs, the multi room services are all still active and producing logs on the device and the device's name is unchanged even after a reboot. I searched the code for use of these variables and didn't find anything.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a Raspberry Pi 3 according to the balenaSound instructions
2. Plug the device in to speakers and make sure it works
3. Set device name and disable multi room variables according to docs
4. Reboot
5. Observe that the device name does not change and multi room services are still producing logs which indicate they are working

**Expected behavior**
I expected the device's bluetooth name to change and for the multi room services to either stop running or to be entirely silent in the console

**Device information**
 - Board (hardware info): Raspberry Pi 3
 - Device type (as seen in balenaCloud): Raspberry Pi 3
 - balenaOS version: balenaOS v2.56.0+rev2

ChestersGarage, 2020-09-16:
@rpecka The environment variables all changed in the last couple days, with the release of version 3.0.  See latest docs on Github or  your local copy of the git repo, if you cloned it.

https://github.com/balenalabs/balena-sound/blob/master/docs/03-customization.md

The new variable names, which replace those you are trying to set are:
SOUND_DEVICE_NAME = <yourdevicename>  (replaces DEVICE_NAME)
SOUND_MODE = <one of MULTI_ROOM, MULTI_ROOM_CLIENT or STANDALONE>  (replaces DISABLE_MULTI_ROOM)

tmigone, 2020-09-16:
Hi @rpecka, as @ChestersGarage is saying, we released yesterday a major update of this project. Unfortunately we are having trouble with the bot that generates our docs site so it's showing an old version. We hope to solve this soon to avoid further conflict.

tmigone, 2020-09-21:
Bot is now behaving! Closing this issue as docs should be up to date now:  https://sound.balenalabs.io/

closed 2020-09-21 by tmigone