iotsound/iotsound #286: SBC Decoding Error about 3 minutes into use

rpecka, 2020-09-16:
**Describe the problem**
I set up my Raspberry Pi 3 according to the instructions and it works well for about 3 minutes, after which the audio stops entirely. In the logs I see:

```
16.09.20 11:53:30 (-0400)  audio  E: [bluetooth] a2dp-codec-sbc.c: Unsupported fragmented SBC frame
16.09.20 11:53:30 (-0400)  audio  E: [bluetooth] module-bluez5-device.c: Decoding error
```

This is similar to the error described under `Audio cuts or is very stuttery` in the troubleshooting guide but not the same. Sometimes it says this `Unsupported fragmented SBC frame` but sometimes I get `SBC decoding error (-2)` instead.

Is this the same issue, which I might be able to solve by buying a USB Bluetooth dongle or is this something else?

**Device information**
 - Board (hardware info): Raspberry Pi 3
 - Device type (as seen in balenaCloud): Raspberry Pi 3
 - balenaOS version: balenaOS v2.56.0+rev2

sylqvq, 2020-09-22:
+1, I'm having this exact same issue, but instead of getting "Unsupported fragmented SBC frame," I get not much more information.

`22.09.20 18:19:15 (-0500)  audio  E: [bluetooth] a2dp-codec-sbc.c: SBC decoding error (-2)
22.09.20 18:19:15 (-0500)  audio  E: [bluetooth] module-bluez5-device.c: Decoding error`

SimonMellerin, 2020-10-09:
Hi,

I've the same problem with :
- a RPI3 
- with or without a bluetooth dongle

There's no problem when I try to play music with Spotify connect, but music stop after 1sec when I try via bluetooth with android phone (I've test with 2 phones, same result each time)

```
09.10.20 13:06:48 (+0200)  audio  W: [pulseaudio] module-loopback.c: Cannot set requested source latency of 66.67 ms, adjusting to 250.00 ms
09.10.20 13:06:56 (+0200)  audio  E: [bluetooth] a2dp-codec-sbc.c: SBC decoding error (-2)
09.10.20 13:06:56 (+0200)  audio  E: [bluetooth] module-bluez5-device.c: Decoding error
09.10.20 13:06:59 (+0200)  audio  W: [pulseaudio] module-loopback.c: Cannot set requested source latency of 66.67 ms, adjusting to 68.54 ms
09.10.20 13:07:00 (+0200)  audio  E: [bluetooth] a2dp-codec-sbc.c: SBC decoding error (-2)
09.10.20 13:07:00 (+0200)  audio  E: [bluetooth] module-bluez5-device.c: Decoding error
```

AlexProgrammerDE, 2020-12-29:
I think this is a duplicate of #62

closed 2020-12-29 by AlexProgrammerDE

fgebhart, 2022-01-06:
For people coming to this issue via google searching the error message:

I managed to solve this issue by using a usb bluetooth dongle.

Using RPI 3, balenaOS 2.83.21+rev1 and Supervisor version 12.11.22.

thermomonkey, 2022-01-10:
> For people coming to this issue via google searching the error message:
> 
> I managed to solve this issue by using a usb bluetooth dongle.
> 
> Using RPI 3, balenaOS 2.83.21+rev1 and Supervisor version 12.11.22.

Here from Google, thanks mate.  This issue has been driving me mad for hours.  Will it just recoganise the bluetooth adapter automatically?

fgebhart, 2022-01-11:
@thermomonkey According to the balena docs it would automatically recognize and use the connected bluetooth dongle.

Turned out I had to put it into a different USB slot and reboot the RPI, but then it automatically used the bluetooth dongle as the new default and disabled the on-board bluetooth.

kifeo, 2024-01-07:
I have the same issue.
I see this issue is closed but is not fixed.
Please reopen to fix this.

providing a workaround does not mean the issue is fixed.
Thanks !