iotsound/iotsound #287: 404 errors among v3.0 documentation links

ChestersGarage, 2020-09-16:
**Describe the problem**
Almost all of the links within the documentation are leading to 404 errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://github.com/balenalabs/balena-sound/blob/master/docs/03-customization.md#general'
2. Click on the "here" link in the Description field for SOUND_MODE.
3. 404 error

Also, the Options field for SOUND_MODE list the following: "MULTI_ROOM, MULTI_ROOM, STANDALONE", which should actually be "MULTI_ROOM, MULTI_ROOM_CLIENT, STANDALONE"

**Expected behavior**
Not a 404 error.

**Additional context**
I'm seeing the same 404 errors in almost all inter-doc links.

tmigone, 2020-09-16:
Hey @ChestersGarage, thanks for the report. The reason the links are 404'ing is that for our docs we use a bot that statically generates a website (here https://sound.balenalabs.io/). It's currently buggy so it hasn't updated the docs for v3.0, but we are working on solving that.

tmigone, 2020-09-21:
Closing this as our bot is now behaving :)
Reminder that the best way to browse docs is https://sound.balenalabs.io/

closed 2020-09-21 by tmigone

ChestersGarage, 2020-09-21:
@tmigone All of the links I noticed originally are still returning 404 errors.  It's all the links that send you to the github documentation.  The files under /docs were renamed, but none of the links were updated to reflect the new names.  

E.g.: Link refs "device-support", but the file is named "04-device-support.md" now.

tmigone, 2020-09-22:
hey @ChestersGarage, yes, that's correct. This is because we "optimize" the links/browsing for our docs website (https://sound.balenalabs.io/). Unfortunately there is no way of having linking work both for GitHub browsing and for our docs website, and we prefer the latter. I'll raise this to the bot team and see if we can get something done so that links are resolved correctly in both cases. Thanks for bringing it up