iotsound/iotsound #288: Ability to disable automatic master/client renegotiation

ChestersGarage, 2020-09-16:
**Is your feature request related to a problem? Please describe.**
I would like to manually manage what clients are associated with each master on my fleet.  This allows me to have multiple different source streams running on different masters, and assigned to specific clients.  A standalone snapcast client is capable of assigning clients to servers manually.  But the automatic renegotiation of the masters foils those capabilities, and prevents multiple server streams.

**Describe the solution you'd like**
I'd like an environment variable that disables auto-renegotiation of masters and clients, so that I can manually assign clients to each master through the snapcast client.

**Additional context**
This would need to be an advanced feature, since the snapcast client is not easily obtained on all platforms, and is a rough UI, not meant for general usage.  The only platform that has an easy solution is Android, where the snapcast client is available as an app. 

[image]

[image]

[image]

tmigone, 2021-06-15:
Hey @ChestersGarage, I just implemented an option that I think would work for this use case. Setting `SOUND_MULTIROOM_DISALLOW_UPDATES` will prevent a device from reconfiguring it's multiroom master whenever it receives an update from the fleet. Let me know if this helps!

closed 2021-06-21 by bulldozer-balena[bot]

ChestersGarage, 2021-06-24:
@tmigone I just saw this.  I'll try it out and see how it goes.

commit a6d7a114 (2021-06-15): multiroom: allow disabling multiroom master updates

Closes: #288
Change-type: patch
commit 035bba04 (2021-06-15): multiroom: allow disabling multiroom master updates

Closes: #288
Change-type: patch
commit 63965d6c (2021-06-15): multiroom: allow disabling multiroom master updates

Closes: #288
Change-type: patch
commit 9d02cd37 (2021-06-18): multiroom: allow disabling multiroom master updates

Closes: #288
Change-type: patch