iotsound/iotsound #289: Deploying master branch to rpi3 application fails

pletnes, 2020-09-16:
**Describe the problem**
A clear and concise description of what the problem is.

**To Reproduce**
Steps to reproduce the behavior:
1. git clone balena-sound repo from github
2. git remote add balena  g_paul_anton_letnes@git.balena-cloud.com:g_paul_anton_letnes/sverresborg.git (fetch)
3. git push balena 
4. See error

**Expected behavior**
Build successful and app deployed.

**Screenshots**
Bottom of logs:
```
[Info]              Uploading images
[Success]           Successfully uploaded images
[Error]             Some services failed to build:
[Error]               Service: audio
[Error]                 Error: manifest for balenaplayground/balenalabs-audio:raspberrypi3-64 not found
[Error]               Service: bluetooth
[Error]                 Error: manifest for balenaplayground/balenalabs-bluetooth:raspberrypi3-64 not found
[Info]              Built on arm01
[Error]             Not deploying release.

remote: error: hook declined to update refs/heads/master
To git.balena-cloud.com:g_paul_anton_letnes/sverresborg.git
 ! [remote rejected] master -> master (hook declined)
error: failed to push some refs to 'git.balena-cloud.com:g_paul_anton_letnes/sverresborg.git'
```

**Device information**
 - Board (hardware info): Raspberry Pi 3B (not sure if + or not)
 - Device type (as seen in balenaCloud):  Raspberry Pi 3 (using 64bit OS)
 - balenaOS version: balenaOS 2.47.0+rev1 (in production mode)

**Additional context**
Can't think of anything.

pletnes, 2020-09-16:
Hi, tried to pull off this cool little project for my home office. Got stuck with "manifest for balenaplayground/balenalabs-audo:raspberrypi3-64 not found" - could it be a docker image missing arm64 support, for instance?

Cheers
Paul Anton

tmigone, 2020-09-16:
Hi @pletnes thanks for letting us know. Indeed that particular case slipped by! I'll be building the missing image shortly.

ChestersGarage, 2020-09-16:
@pletnes Per the device support doc [1] balena-sound does not support 64-bit OS on RPi 3 and 4.  Need to use a 32-bit OS image. 

[1] https://github.com/balenalabs/balena-sound/blob/master/docs/04-device-support.md#recommended  (see note 3)

@tmigone Do you expect v3.x to be compatible with 64-bit OS? That would be very cool!  :)

tmigone, 2020-09-16:
@ChestersGarage I think that's badly worded tbh, needs to be corrected. It should work though you might need to apply the workaround described here: https://sound.balenalabs.io/docs/support#troubleshooting (scroll down to the issue `No audio when using balenaOS 64 bit on Raspberry Pi 3's`)

pletnes, 2020-09-17:
I tried to use the big, friendly "deploy with balena" on this blogpost. This created a new application in my account.
https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/
Now my RPi can play audio to a USB sound card, so this is working fine. The OS is still 64-bit, as I moved the same rpi to the new application.

Thoughts? Why is this working?

pletnes, 2020-09-17:
If anyone is working at balena and want more details / support access, let me know.

tmigone, 2020-09-17:
Hi @pletnes, not sure I follow. Why is what working?

pletnes, 2020-09-17:
There's a big, friendly "deploy wth balena" in the blog post. I clicked the button and followed the flow. Now I've got balena sound working, on a 64 bit OS. It was stated above that balena sound only works on a 32 bit balena OS. It sure seems like it's working for me.

So @ChestersGarage said balena-sound does not work on 64 bit balenaOS, but it now works for me, without any workarounds, if I use the "deploy" button.

Just an observation.

AlexProgrammerDE, 2020-09-17:
@pletnes as Tomas told you:
> I think that's badly worded tbh, needs to be corrected.
He means that there were issues with 64 bit os spotted, but a workaround was found. So don't worry about it.
He also said:
> I'll be building the missing image shortly.
Seems like he builded them. And now its working perfectly. 
Btw we are moving the repository for the images soon. 
I will close this issue for now.

closed 2020-09-17 by AlexProgrammerDE

pletnes, 2020-09-17:
Right, now I get what you're all saying. Thanks, very quick response!

tmigone, 2020-09-17:
To further clarify what Alex said:

- balenaSound should work with 64 bit versions of balenaOS, both for the Pi3 and Pi4 and both for balenaSound v2.x and v3.x
- For the Pi3 64 bit, you need to remove a dtoverlay to get audio out of the audio jack/HDMI. You don't need to do this if you use a DAC or USB soundcards (@pletnes this is the reason yours worked out of the box).

About the missing docker images, I've built and published the missing Pi3-64 images. I've also switched the project to use our production dockerhub (`balenablocks`) instead of the staging env (`balenaplayground`). This applies to the latest version onwards. Regardless, the missing images should be published and available in both cases.

Let me know if you guys have more questions! And thanks for trying out balenaSound!

tmigone, 2020-09-17:
We do need to make that section on the docs more clear, so thanks for bringing that up!

AlexProgrammerDE, 2020-09-17:
@tmigone imma make a commit for that.