iotsound/iotsound #294: Audio stuttering & PulseAudio "complaning"

b3nis, 2020-09-17:
**Describe the problem**
Audio is stuttering and PulseAudio is reporting a lof of underruns. Only using one Pi 4 (2GB) (server and client on same device).

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy a fresh BalenaSound image to a Pi 4
2. Boot it up
3. Let it update
4. Play music via Spotify Connect

**Expected behavior**
No audio stuttering.

**Screenshots**
[image]

**Device information**
 - Board (hardware info): Raspberry Pi 4 (2GB)
 - Device type (as seen in balenaCloud): I must be blind because I cannot see that...
 - balenaOS version: balenaOS 2.56.0+rev1

AlexProgrammerDE, 2020-09-18:
@b3nis hm this sounds weird. Can you try setting it to Standalone mode?

duchu, 2020-09-19:
@AlexProgrammerDE i have similar problem but for me there is no any audio. How i can switch to Standalone mode?

tmigone, 2020-09-22:
Hi @b3nis @duchu, switching to standalone won't change much. These underruns happen because PulseAudio starts with an aggressively short playback latency and increments it as needed. In my experience, it can happen quite often (as seen on the screenshot) when you start playback but will quickly stabilize and stop happening. Has this been your experience too?

Regardless, I'm investigating how we can be more lenient with the initial latency to avoid this problem. I've got a couple of avenues worth exploring but it might take a few days.

tmigone, 2020-10-28:
A couple users reported switching to `STANDALONE` mode does help. This warrants further investigation, I'll try get into it soon.

Also copying the warning message in plain text so it's easier to find it using search:

```
W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 320.00 ms
```

xvisor16, 2020-11-05:
I am experiencing the same issues on a RPi 3B. But sometimes I see the underrun message when it happens but sometimes I see different messages:
05.11.20 16:56:59 (+0100)  audio  W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 340.00 ms
05.11.20 16:57:11 (+0100)  multiroom-server  2020-11-05 15-57-11 [Info] (onResync) onResync (balenaSound): 87 ms
05.11.20 16:57:12 (+0100)  multiroom-server  2020-11-05 15-57-12 [Info] (do_read) next read < 0 (balenaSound): -0.596 ms
05.11.20 16:57:12 (+0100)  multiroom-server  2020-11-05 15-57-12 [Info] (onResync) onResync (balenaSound): 118 ms
05.11.20 16:57:13 (+0100)  multiroom-client  2020-11-05 15-57-13 [Info] (Stream) pMiniBuffer->full() && (abs(pMiniBuffer->mean()) > 50): -161896
05.11.20 16:57:16 (+0100)  multiroom-server  2020-11-05 15-57-16 [Info] (do_read) next read < 0 (balenaSound): -21.858 ms
05.11.20 16:57:16 (+0100)  multiroom-server  2020-11-05 15-57-16 [Info] (do_read) next read < 0 (balenaSound): -3.601 ms
05.11.20 16:57:17 (+0100)  multiroom-server  2020-11-05 15-57-17 [Info] (onResync) onResync (balenaSound): 254 ms
05.11.20 16:57:17 (+0100)  multiroom-server  2020-11-05 15-57-17 [Info] (onResync) onResync (balenaSound): 165 ms
05.11.20 16:57:17 (+0100)  multiroom-server  2020-11-05 15-57-17 [Info] (do_read) next read < 0 (balenaSound): -23.261 ms
05.11.20 16:57:17 (+0100)  multiroom-server  2020-11-05 15-57-17 [Info] (do_read) next read < 0 (balenaSound): -4.727 ms
05.11.20 16:57:17 (+0100)  multiroom-server  2020-11-05 15-57-17 [Info] (onResync) onResync (balenaSound): 59 ms
05.11.20 16:57:17 (+0100)  multiroom-client  2020-11-05 15-57-17 [Info] (Stream) abs(age > 500): 511300
05.11.20 16:57:17 (+0100)  multiroom-client  2020-11-05 15-57-17 [Info] (Alsa) Failed to get chunk
05.11.20 16:57:19 (+0100)  multiroom-client  2020-11-05 15-57-19 [Info] (Stream) pMiniBuffer->full() && (abs(pMiniBuffer->mean()) > 50): -129488

Don't know if that could be related? I have tried DISABLE_MULTI_ROOM = 1 which makes it less but doesn't completely resolve it. Load on the Pi is low, everything but memory is green in Balena but memory, max load on the device comes from pulseaudio which is about 6 a 7 percent on the host OS. In balena total load doesn't reach above 15%. Temp is 50c.

tmigone, 2020-11-19:
@xvisor16, @duchu, @b3nis 

If you guys can help me out here I could use some logs with extra verbosity. You'll need to set environment variable `AUDIO_LOG_LEVEL` to `DEBUG` and then share with me the logs. You'll know it's on `DEBUG` mode because logs will show the following:

```
19.11.20 18:34:42 (-0300)  audio  --- Audio ---
19.11.20 18:34:42 (-0300)  audio  Starting audio service with settings:
19.11.20 18:34:42 (-0300)  audio  - Pulse log level: DEBUG
```

What I'm looking for in the logs is instances of this error occurring, so make sure you get a few:

```
W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 320.00 ms
```

Thanks!

xvisor16, 2020-11-19:
@tmigone 
Here you go!

[balena-sound2_white-snow-19.11.20_23_27_54_(+0100).txt](https://github.com/balenalabs/balena-sound/files/5570093/balena-sound2_white-snow-19.11.20_23_27_54_.%2B0100.txt)

tmigone, 2020-11-20:
Thanks @xvisor16, for the time being this should suffice. I'll let you know if I need access. Thank you.

tmigone, 2020-11-23:
@xvisor16, @duchu, @b3nis, @ChestersGarage hey there. 

I just opened PR [#356](https://github.com/balenalabs/balena-sound/pull/356) which contains changes that should fix or at the very least *drastically* improve the audio stuttering related to the message:

```
Too many underruns, increasing latency to xxx.00 ms
```

Changes made are the following:

1) Significantly reduce the CPU and I/O usage which were the biggest contributors to this problem, specially for multiroom scenarios. This probably improves performance on lesser devices too (Pi0 for instance). In my case, running multiroom on Pi 4's would always start off and increase the latency all the way up to 270msec, but with this update it has never complained and is also smoother. 

2) Introduces a new environment variable `SOUND_INPUT_LATENCY` that allows you to directly set the latency that's causing the problem. If you are still having problems you can increase the base latency with this variable. You should let it run for a while until it stabilises and use whatever end value you found. This way the audio server won't go from 200msec all the way up to whatever value in 5msec increments but rather start at the end value. Downside is that you are adding a latency to the pipeline ofc.

As you can see, 1) is the actual fix while 2) is just an additional workaround. My hope is, and the limited testing I've done is in line with it, that you shouldn't need to set the `SOUND_INPUT_LATENCY` at all.

If you are interesting in more details about this you can check the following:
- https://github.com/balenablocks/audio/issues/44
- https://github.com/balenalabs/balena-sound/pull/356

Now the big ask :D I'd love it if you could try it and let me know how it goes! You'll need to deploy the application using the balena CLI or use this handy dandy DWB I created just for this purpose:

[image]

Thanks!

xvisor16, 2020-11-25:
@tmigone Hi there, well one simple but satisfactory answer, YES! You fixed the issue, my Pi3 doesn't even up the latency at all, not even once! Thanks, now I can really start building my complete setup.

By the way, could I also experiment by setting the sound_input_latency even lower? Maybe like 50ms so it's more responsive??

xvisor16, 2020-11-25:
Aw well I already tried, and found that's not possible...

25.11.20 02:47:07 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 10.64 ms + 39.93 ms + -0.48 ms = 50.03 ms
25.11.20 02:47:07 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 50.03 ms
25.11.20 02:47:07 (+0100)  audio  D: [pulseaudio] module-loopback.c: [snapcast] Updated sampling rate to 44100 Hz.
25.11.20 02:47:07 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 64.65 ms + 145.87 ms + -9.67 ms = 200.65 ms
25.11.20 02:47:07 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.65 ms
25.11.20 02:47:07 (+0100)  audio  D: [pulseaudio] module-loopback.c: [alsa_output.dac.stereo-fallback] Updated sampling rate to 44102 Hz.
25.11.20 02:47:17 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 18.82 ms + 35.06 ms + -3.76 ms = 50.03 ms
25.11.20 02:47:17 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 50.03 ms
25.11.20 02:47:17 (+0100)  audio  D: [pulseaudio] module-loopback.c: [snapcast] Updated sampling rate to 44100 Hz.
25.11.20 02:47:17 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 45.43 ms + 163.85 ms + -8.44 ms = 200.74 ms
25.11.20 02:47:17 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.75 ms
25.11.20 02:47:17 (+0100)  audio  D: [pulseaudio] module-loopback.c: [alsa_output.dac.stereo-fallback] Updated sampling rate to 44103 Hz.
25.11.20 02:47:27 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 13.75 ms + 37.12 ms + -0.79 ms = 50.03 ms
25.11.20 02:47:27 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 50.03 ms
25.11.20 02:47:27 (+0100)  audio  D: [pulseaudio] module-loopback.c: [snapcast] Updated sampling rate to 44100 Hz.
25.11.20 02:47:27 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 65.86 ms + 154.41 ms + -19.61 ms = 200.59 ms
25.11.20 02:47:27 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.60 ms

But I am happy anyway!

tmigone, 2020-11-27:
Hey @xvisor16, glad this makes things better for you!! 💪 💪 💪 💪 
In theory you can go with a latency as low as 1msec. In reality as you found out, the latency you configure is merely a `suggestion` made to pulseaudio, and it will increase if it's not possible to satisfy it.

I'll merge the PR soon so that it hits the master branch as I've seen mainly great feedback about it. Thanks again for helping out!

closed 2020-11-27 by bulldozer-balena[bot]

xvisor16, 2020-11-28:
Off course no problem all! Thank you for fixing it! Which is way more important 😉Cheers, Peter -------- Oorspronkelijk bericht --------Van: Tomás Migone <notifications@github.com>Datum: vr 27 nov. 2020 19:07Aan: balenalabs/balena-sound <balena-sound@noreply.github.com>Cc: xvisor16 <github@etict.nl>, Mention <mention@noreply.github.com>Onderwerp: Re: [balenalabs/balena-sound] Audio stuttering & PulseAudio "complaning" (#294)
Hey @xvisor16, glad this makes things better for you!! 💪 💪 💪 💪
In theory you can go with a latency as low as 1msec. In reality as you found out, the latency you configure is merely a suggestion made to pulseaudio, and it will increase if it's not possible to satisfy it.
I'll merge the PR soon so that it hits the master branch as I've seen mainly great feedback about it. Thanks again for helping out!

heyarne, 2021-03-13:
I am using the my balena sound device as a pulseaudio tunnel sink as described in #422.
I am still quite frequently experiencing stuttering, at least once every 5 minutes or so. The relevant log entries:

<details>
<summary>Log output</summary>

```
13.03.21 19:53:32 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.16 ms
13.03.21 19:53:32 (+0100)  audio  D: [pulseaudio] module-loopback.c: [balena-sound.output] Updated sampling rate to 44100 Hz.
13.03.21 19:53:42 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 59.80 ms + 207.87 ms + -67.27 ms = 200.32 ms
13.03.21 19:53:42 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.32 ms
13.03.21 19:53:42 (+0100)  audio  D: [pulseaudio] module-loopback.c: [alsa_output.dac.stereo-fallback] Updated sampling rate to 44101 Hz.
13.03.21 19:53:42 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 63.02 ms + 150.02 ms + -12.76 ms = 200.16 ms
13.03.21 19:53:42 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.16 ms
13.03.21 19:53:42 (+0100)  audio  D: [pulseaudio] module-loopback.c: [balena-sound.output] Updated sampling rate to 44100 Hz.
13.03.21 19:53:52 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 65.10 ms + 177.07 ms + -41.83 ms = 200.22 ms
13.03.21 19:53:52 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.22 ms
13.03.21 19:53:52 (+0100)  audio  D: [pulseaudio] module-loopback.c: [alsa_output.dac.stereo-fallback] Updated sampling rate to 44100 Hz.
13.03.21 19:53:52 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 39.11 ms + 187.53 ms + -26.34 ms = 200.16 ms
13.03.21 19:53:52 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.16 ms
13.03.21 19:53:52 (+0100)  audio  D: [pulseaudio] module-loopback.c: [balena-sound.output] Updated sampling rate to 44100 Hz.
13.03.21 19:54:02 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 58.73 ms + 161.11 ms + -19.39 ms = 200.33 ms
13.03.21 19:54:02 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.33 ms
13.03.21 19:54:02 (+0100)  audio  D: [pulseaudio] module-loopback.c: [alsa_output.dac.stereo-fallback] Updated sampling rate to 44101 Hz.
13.03.21 19:54:02 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 15.26 ms + 187.53 ms + -2.52 ms = 200.16 ms
13.03.21 19:54:02 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.16 ms
13.03.21 19:54:02 (+0100)  audio  D: [pulseaudio] module-loopback.c: [balena-sound.output] Updated sampling rate to 44100 Hz.
13.03.21 19:54:12 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 63.26 ms + 207.68 ms + -70.61 ms = 200.23 ms
13.03.21 19:54:12 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.24 ms
13.03.21 19:54:12 (+0100)  audio  D: [pulseaudio] module-loopback.c: [alsa_output.dac.stereo-fallback] Updated sampling rate to 44101 Hz.
13.03.21 19:54:12 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 66.45 ms + 150.02 ms + -16.21 ms = 200.16 ms
13.03.21 19:54:12 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.16 ms
13.03.21 19:54:12 (+0100)  audio  D: [pulseaudio] module-loopback.c: [balena-sound.output] Updated sampling rate to 44100 Hz.
13.03.21 19:54:22 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 65.24 ms + 180.49 ms + -45.49 ms = 200.13 ms
13.03.21 19:54:22 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.13 ms
13.03.21 19:54:22 (+0100)  audio  D: [pulseaudio] module-loopback.c: [alsa_output.dac.stereo-fallback] Updated sampling rate to 44100 Hz.
13.03.21 19:54:22 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 42.58 ms + 187.53 ms + -29.83 ms = 200.16 ms
13.03.21 19:54:22 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.16 ms
13.03.21 19:54:22 (+0100)  audio  D: [pulseaudio] module-loopback.c: [balena-sound.output] Updated sampling rate to 44100 Hz.
13.03.21 19:54:32 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 62.55 ms + 160.84 ms + -23.04 ms = 200.24 ms
13.03.21 19:54:32 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.24 ms
13.03.21 19:54:32 (+0100)  audio  D: [pulseaudio] module-loopback.c: [alsa_output.dac.stereo-fallback] Updated sampling rate to 44101 Hz.
13.03.21 19:54:32 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 18.75 ms + 187.53 ms + -6.01 ms = 200.16 ms
13.03.21 19:54:32 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.16 ms
13.03.21 19:54:32 (+0100)  audio  D: [pulseaudio] module-loopback.c: [balena-sound.output] Updated sampling rate to 44100 Hz.
13.03.21 19:54:42 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 65.16 ms + 207.86 ms + -72.80 ms = 200.12 ms
13.03.21 19:54:42 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.12 ms
[... cut ...]
13.03.21 19:55:09 (+0100)  audio  D: [null-sink] source.c: Processing rewind...
13.03.21 19:55:09 (+0100)  audio  D: [null-sink] source-output.c: Have to rewind 4972 bytes on implementor.
13.03.21 19:55:09 (+0100)  audio  D: [null-sink] module-null-sink.c: Rewound 4972 bytes.
13.03.21 19:55:12 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback overall latency is 54.39 ms + 161.16 ms + -15.15 ms = 200.31 ms
13.03.21 19:55:12 (+0100)  audio  D: [pulseaudio] module-loopback.c: Loopback latency at base rate is 200.31 ms
```
</details>

dolfje, 2021-04-11:
Same here, lot of lag. So if press play, or move the volume higher on my bluetooth android, it takes more than 5s to respond. Raspberry pi zero w + hifiberry amp2. Also a lot of stutter, so repeats of the same phrase and cracks (music ringbuffer not fast enough filled?).  Also a lot of "increasing latency". Just installed yesterday, so I guess the newest version?

I have tried Pi MusicBox, almost no stutter, but it is still there. And latency is also 5s+.
I have tried moode player, no stutter + latency 500ms

So at the moment with moode player, but willing to do experiments...

commit 095b56ca (2020-11-23): multiroom: update snapcast to v0.22 and various improvements

- Update snapcast to v0.22.0-r0
- Replace FIFO stream source with ALSA which significantly reduces CPU and I/O usage (fixes #294)
- Reduce multiroom-server image with multi stage builds for a slimmer image
- Reduce verbose level for both multiroom services

Connects-to: #294
Change-type: minor
commit 7dd06fc7 (2020-11-23): multiroom: update snapcast to v0.22 and various improvements

- Update snapcast to v0.22.0-r0
- Replace FIFO stream source with ALSA which significantly reduces CPU and I/O usage (fixes #294)
- Add SOUND_INPUT_LATENCY and SOUND_OUTPUT_LATENCY env vars (also help with #294)
- Reduce multiroom-server image with multi stage builds for a slimmer image
- Reduce verbose level for both multiroom services

Connects-to: #294
Change-type: minor
commit 4ff6e6e2 (2020-11-23): multiroom: update snapcast to v0.22 and various improvements

- Update snapcast to v0.22.0-r0
- Replace FIFO stream source with ALSA which significantly reduces CPU and I/O usage (fixes #294)
- Add SOUND_INPUT_LATENCY and SOUND_OUTPUT_LATENCY env vars (also helps with #294)
- Reduce multiroom-server image with multi stage builds for a slimmer image
- Reduce verbose level for both multiroom services

Connects-to: #294
Change-type: minor
commit b7ba5b59 (2020-11-23): multiroom: update snapcast to v0.22 and various improvements

- Update snapcast to v0.22.0-r0
- Replace FIFO stream source with ALSA which significantly reduces CPU and I/O usage (fixes #294)
- Add SOUND_INPUT_LATENCY and SOUND_OUTPUT_LATENCY env vars (also helps with #294)
- Reduce multiroom-server image with multi stage builds for a slimmer image
- Reduce verbose level for both multiroom services

Connects-to: #294
Change-type: minor