iotsound/iotsound #295: Balena Sound as a Service/Application

salsabeard, 2020-09-18:
I'm guessing I'm out of my depths in asking this question, but would it ever be possible to have Balena Sound offered as a service or application for Windows, Mac, and/or Linux?  I've just been thinking that it would be awesome to have my Logitech Z906 system on my PC tapped into the multi-room.

I'm not thinking from the perspective of the PC sending all audio out to all of the multi-room speakers, but rather to have an application available as a receiver that can play through to the speakers.  So say for instance I'm sitting at my computer working and listening to Spotify, I could change the audio from "This Computer" to "This Computer + Multi" (or something like that).  It wouldn't switch all of the PC's sound to multi-room, but rather shift Spotify to play from that service/application that was connected to the rest of the grid.

At the moment, the only idea I've come up with is running a RPi as a VM on my PC and deploying my Balena Sound Application  to it.  Just wondering if there's a better way.

tmigone, 2020-09-22:
Hey @salsabeard, I get what you say. Unfortunately balenaSound is tightly coupled to balenaOS and the Linux kernel. The VM approach might work, maybe not all functionality, if you use a linux based machine to host it. 
It's a cool idea so i'll keep it in mind and see if I can think of a better solution.

tmigone, 2021-04-20:
Closing this issue as balenaSound is increasingly getting coupled with balenaOS features and I don't see us forking a non-balena version.

closed 2021-04-20 by tmigone