iotsound/iotsound #296: Update to 3.0 not working with RPi3b Suptronics x400 DAC

peteretep, 2020-09-18:
**Describe the problem**
I  have been using balena sound with a Raspberry Pi3b and a Suptronics x400 DAC.
It worked fine up to the update to version 3.
I'm not getting any sound now.
The logs indicate that the spotify plugin is playing the chosen song.

**Device information**
 - Raspberry Pi 3B+ 
- Suptronics x400 DAC
 - Device type: Raspberry Pi3
 - balenaOS version:balenaOS 2.56.0+rev2

**Additional context**
balena sound 3 is working on my pi zero with pimoroni DAC

tmigone, 2020-09-22:
hey @peteretep, can you check the value of your `RESIN_HOST_CONFIG_dtparam/BALENA_HOST_CONFIG_dtparam`. For 2.x you needed to set `”audio=off”` but for 3.0 you don't, so I suspect that might be the reason (maybe).

peteretep, 2020-09-23:
Thanks @tmigone, I checked this again and it has not helped. 
[image]

Have tried setting `audio=on` and also removing the audio setting altogether

tmigone, 2020-09-23:
Ok that's strange! Does the audio jack output work?

peteretep, 2020-09-23:
no it doesn't seem to

peteretep, 2020-09-24:
similar to #306, I am getting these messages:

24.09.20 10:19:32 (+0100)  audio  E: [pulseaudio] module-loopback.c: No such sink.
24.09.20 10:19:32 (+0100)  audio  E: [pulseaudio] module.c: Failed to load module "module-loopback" (argument: "source="balena-sound.output.monitor" sink="alsa_output.dac.stereo-fallback""): initialization failed.
24.09.20 10:19:32 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.

adr1anh, 2020-09-24:
I have been getting the same issue. 
Device information

- Raspberry Pi 3B+
- Hifiberry DAC +
- Device type: Raspberry Pi3
- balenaOS version:balenaOS balenaOS 2.47.0+rev1 64 bit

Connecting by SSH to the `audio` service, I was able to test using `speaker-test -c2 -D plughw:0,0`. This works as expected.

Interestingly, even when I set the device variable `SOUND_OUTPUT` to `RPI_HEADPHONES`, the same error occurs and sound does not work either (I previously set it to `DAC`).

I don't have any dt params set currently.

adr1anh, 2020-09-24:
> hey @peteretep, does the `audio` container start? If so, can you share the output of this command: `pactl list sinks`

Here's the output from my device, where I was having the same error message and symptoms:
```
Sink #0
        State: SUSPENDED
        Name: alsa_output.0.stereo-fallback
        Description: Built-in Audio Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 6
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.0.stereo-fallback.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY 
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = ""
                alsa.id = "HiFiBerry DAC+ HiFi pcm512x-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "snd_rpi_hifiberry_dacplus"
                alsa.long_card_name = "snd_rpi_hifiberry_dacplus"
                alsa.driver_name = "snd_soc_hifiberry_dacplus"
                device.bus_path = "/devices/platform/soc/soc:sound/sound/card0"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card0"
                device.form_factor = "internal"
                device.string = "hw:0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.profile.name = "stereo-fallback"
                device.profile.description = "Stereo"
                device.description = "Built-in Audio Stereo"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Ports:
                analog-output: Analog Output (priority: 9900)
        Active Port: analog-output
        Formats:
                pcm

Sink #1
        State: IDLE
        Name: balena-sound.input
        Description: Null Output
        Driver: module-null-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 22
        Mute: no
        Volume: front-left: 49152 /  75% / -7.50 dB,   front-right: 49152 /  75% / -7.50 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: balena-sound.input.monitor
        Latency: 23528 usec, configured 66666 usec
        Flags: DECIBEL_VOLUME LATENCY 
        Properties:
                device.description = "Null Output"
                device.class = "abstract"
                device.icon_name = "audio-card"
        Formats:
                pcm

Sink #2
        State: RUNNING
        Name: balena-sound.output
        Description: Null Output
        Driver: module-null-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 23
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: balena-sound.output.monitor
        Latency: 23465 usec, configured 66666 usec
        Flags: DECIBEL_VOLUME LATENCY 
        Properties:
                device.description = "Null Output"
                device.class = "abstract"
                device.icon_name = "audio-card"
        Formats:
                pcm

Sink #3
        State: SUSPENDED
        Name: snapcast
        Description: Unix FIFO sink /var/cache/snapcast/snapfifo
        Driver: module-pipe-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 24
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: snapcast.monitor
        Latency: 0 usec, configured 0 usec
        Flags: DECIBEL_VOLUME LATENCY 
        Properties:
                device.string = "/var/cache/snapcast/snapfifo"
                device.description = "Unix FIFO sink /var/cache/snapcast/snapfifo"
                device.icon_name = "audio-card"
        Formats:
                pcm
```

I can confirm that the `audio` service starts and does not crash.

tmigone, 2020-09-24:
Thanks @adr1anh, that's very helpful. The DAC is being detected as `alsa_output.0.stereo-fallback` instead of `alsa_output.dac.stereo-fallback` so at least we now know what is going on. We have some custom UDEV rules to handle the naming conventions, so I'll need to look at that and see why it's failing. I'll get back to you in a few.

tmigone, 2020-09-24:
Got one more request, can you provide the output of `http://<DEVICE_IP>:3000/support`? Need to hit that with your browser, replace `<DEVICE_IP>` with your devices' IP address.

adr1anh, 2020-09-24:
Thank you so much for your help 😊 

Here's the output.

```
{
   "config":{
      "mode":"STANDALONE",
      "device":{
         "ip":"192.168.2.241",
         "type":"raspberrypi3-64"
      },
      "multiroom":{
         "master":"192.168.2.241",
         "forced":false
      },
      "audioBlock":{
         "_events":{
            
         },
         "_eventsCount":2,
         "pulseCookie":{
            "type":"Buffer",
            "data":[
               0,
               66,
               81,
               194,
               85,
               0,
               0,
               0,
               110,
               9,
               0,
               0,
               169,
               1,
               0,
               0,
               152,
               65,
               56,
               194,
               85,
               0,
               0,
               0,
               111,
               9,
               0,
               0,
               44,
               0,
               0,
               0,
               192,
               65,
               81,
               194,
               85,
               0,
               0,
               0,
               216,
               65,
               81,
               194,
               85,
               0,
[... 647 lines cut ...]
            [
               "device.icon_name",
               "audio-card"
            ]
         ],
         "cardIndex":"0n",
         "ports":65536,
         "activePortName":2,
         "formats":65537
      }
   ],
   "volume":75,
   "constants":{
      "debug":false,
      "port":3000,
      "coteDelay":5000,
      "mode":"STANDALONE",
      "balenaDeviceType":"raspberrypi3-64",
      "multiroom":{
         "forced":false
      },
      "volume":75
   }
}
```

tmigone, 2020-09-24:
Ok thanks so much, I think I've figured it out. Going to try a few things and report back.

tmigone, 2020-09-24:
hey @adr1anh @peteretep last request, can you run `dmesg | grep hifi` and `dmesg | grep snd` in the `audio` container and share that? Thanks!

adr1anh, 2020-09-24:
Sure thing:

```
root@0911fe76d695:/usr/src# dmesg | grep hifi
[   16.013916] snd-rpi-hifiberry-dacplus soc:sound: ASoC: CPU DAI (null) not registered - will retry
[   16.045576] snd-rpi-hifiberry-dacplus soc:sound: ASoC: CPU DAI (null) not registered - will retry
[   16.170206] snd-rpi-hifiberry-dacplus soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   16.219356] snd-rpi-hifiberry-dacplus soc:sound: pcm512x-hifi <-> 3f203000.i2s mapping ok
[   16.235456] snd-rpi-hifiberry-dacplus soc:sound: ASoC: no DMI vendor name!
```

```
root@0911fe76d695:/usr/src# dmesg | grep snd
[   16.013916] snd-rpi-hifiberry-dacplus soc:sound: ASoC: CPU DAI (null) not registered - will retry
[   16.045576] snd-rpi-hifiberry-dacplus soc:sound: ASoC: CPU DAI (null) not registered - will retry
[   16.170206] snd-rpi-hifiberry-dacplus soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   16.219356] snd-rpi-hifiberry-dacplus soc:sound: pcm512x-hifi <-> 3f203000.i2s mapping ok
[   16.235456] snd-rpi-hifiberry-dacplus soc:sound: ASoC: no DMI vendor name!
```

tmigone, 2020-09-24:
Fantastic, that settles it. I'll get the fix out, probably won't make it today but tomorrow for sure. Thank you for helping out.

adr1anh, 2020-09-24:
Thank you so much for helping out!

peteretep, 2020-09-24:
Similar output -

```
root@50bd7fb3dcfa:/usr/src# dmesg | grep hifi
[   10.077984] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.147732] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.183667] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.224758] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.364107] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.727790] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.740535] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.779238] snd-rpi-iqaudio-dac soc:sound: pcm512x-hifi <-> 3f203000.i2s mapping ok
```

```
root@50bd7fb3dcfa:/usr/src# dmesg | grep snd
[   10.077984] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.147732] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.183667] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.224758] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.364107] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.727790] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.740535] snd-rpi-iqaudio-dac soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[   10.779238] snd-rpi-iqaudio-dac soc:sound: pcm512x-hifi <-> 3f203000.i2s mapping ok
```

Thanks!

tmigone, 2020-09-25:
Should be fixed by https://github.com/balenablocks/audio/commit/a865d02e7be13a66000cb7eb95c5f63389d54991

Easiest way to update is to re-push balenaSound to your device. Let me know if it works!

adr1anh, 2020-09-25:
Just tested it and it works (even better than before even, I feel).

Thank you so much for getting this done 😊

tmigone, 2020-09-25:
Glad to know! I'll wait for @peteretep to reply before closing this

peteretep, 2020-09-28:
working now, thanks.

closed 2020-09-28 by peteretep