iotsound/iotsound #298: GPIO pause and interupt

pitchuk, 2020-09-19:
**Is your feature request related to a problem? Please describe.**
I work in an electronic lab and we have a number of doors and door bells. Some reach to some rooms others to others, it is a bit of a mess. 

**Describe the solution you'd like**
Is there a way of getting the current thing playing to pause a doorbell sound to play for a bit them resume when a GPIO goes high. Multiple pins have different sounds

**Describe alternatives you've considered**
If i knew there was a way that may work belena_sense + these commands I would have a play

**Additional context**

tmigone, 2020-09-22:
Hey @pitchuk, controlling playback from GPIO in such way is not currently possible, I might look into it though I can't promise at this point when that might happen. 
Would playing the doorbell over the music work? Starting with v3.0, balenaSound can play simultaneous audio sources at the same time. I think it should be fairly straightforward to add one container that checks GPIO and plays back a doorbell sound (without having music stop).

pitchuk, 2020-09-23:
Thanks for your reply! I will await the update, but will look for other options in the meantime

closed 2020-09-23 by pitchuk