iotsound/iotsound #300: Control volume with my phone

N445, 2020-09-22:
**Describe the problem**
I can't control the sound volume with my phone. I have only 2 state volume 0% and volume 100%.
I have an iphone if it can be help you

**To Reproduce**
Steps to reproduce the behavior:
1. Install Balena sound
2. Play music
3. Push the volume button up and down
4. No sound modification

**Expected behavior**
I want to controlle the sound volume with my cellphone (or other devices)

**Screenshots**
No screen shots

**Device information**
 - Board (hardware info): Raspberry pi 4
 - Device type (as seen in balenaCloud): Raspberry pi 4
 - balenaOS version: balenaOS 2.56.0+rev1

tmigone, 2020-09-22:
Hey @N445 is this while streaming via Bluetooth or other service?

N445, 2020-09-22:
Hi ! 
sorry I forgot to specify this : I use bluetooth

tmigone, 2020-09-22:
Thanks for reporting. This indeed should not be the case so I'll take a look in the upcoming days to fix this.

TwoTenPvP, 2020-10-27:
Any updates on this?

xsalon, 2020-11-02:
Same issues with balenaOS 2.58.3+rev1 on RPi3B+ and Samsung Galaxy S10e. (Win10 same behaviour)

ejannett, 2020-11-04:
Hi,
same behaviour with balenaOS 2.48.0+rev1 on Raspberry Pi Zero

N445, 2020-11-04:
It still not working on balenaOS 2.58.6+rev1 , Raspberry pi 4 , iOs 14

ejannett, 2020-11-05:
For you information not working on
raspberry zero running  balenaOS 2.54.2+rev1. and latest balenasound code
iOS 14.1

ejannett, 2020-11-05:
Hi,
googling a little bit about the issue I found a site talking about the same . it seems that we have to force the 
load of the a2dp plugin. doing this on one of my devices worked: volume control is back !

```
# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/etc/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/bluetooth.service.d
           └─bluetooth.conf
   Active: active (running) since Thu 2020-11-05 09:39:09 UTC; 18min ago
     Docs: man:bluetoothd(8)
 Main PID: 1529 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 2300)
   Memory: 1.3M
   CGroup: /system.slice/bluetooth.service
           └─1529 /usr/libexec/bluetooth/bluetoothd **--plugin=a2dp** --experimental

Nov 05 09:39:09 562f511 bluetoothd[1529]: Ignoring (cli) gap

```

eiddor, 2020-11-05:
Good find @ejannett - Can you tell me how/where you added that option?  I know @tmigone is working on the issue per the jellyfish links above.

ejannett, 2020-11-05:
Hi,
Here is what I've done. This is purely testing , proper solution in Balena stock is needed. 
Use at your own risk :-)

```
#log into the device
balena ssh <device id>
mount -o remount,rw /
systemctl edit --full bluetooth # <-- add '--plugin=a2dp' to the  bluetoothd command line. and save the file 
```
restart the device afterwards.

hope it'll work for you 
manu

eiddor, 2020-11-05:
@ejannett This worked perfectly for me - Thanks for the workaround!

Jimbobiss, 2020-12-11:
Any update on this issue? I’m too new to be able to get @ejannett’s fix working, so it’d be great to know what point the fix has reached

eiddor, 2020-12-11:
Hey @Jimbobiss - I'm not sure where they are with the fix, but I can break down @ejannett's fix a bit to maybe make it easier for you to apply if that would help you.

Jimbobiss, 2020-12-11:
Hi there @eiddor. That would be very kind of you if you're willing to do so. I've been using the terminal from the Balena dashboard within my balena-sound application, selecting the Host OS as the target. I'm able to successfully execute the commands @ejannett specified up until 'systemctl edit --full bluetooth'. It returns the bluetooth file for me to edit and I have found I can use 'Insert' to start editing and press 'Esc' to stop editing, but I can't for the life of me figure out how to save my updated file. once I'm in the the context of the file, there doesn't seem to be any way out. Indeed the only way to exit (without closing the terminal instance) I've found is to enter ':q!' and this only seems to kill that particular command process. If you could offer me some guidance on that particular area, I would be most grateful; as I said, I'm extremely new to this sort of method of operation and as a result it's still fairly foreign to me.

eiddor, 2020-12-11:
@Jimbobiss You're 99.9% there.  You're in vi, so the command to quit with a save is ```:wq``` - After that, reset the device from the Balena console and things should work.

The ```ExecStart``` line should look like this:

```
ExecStart=/usr/libexec/bluetooth/bluetoothd --experimental --plugin=a2dp
```

Jimbobiss, 2020-12-11:
@eiddor Thanks for getting back to me - I've tried your suggestion and for the first time I can see that the bluetooth file has updated! Thanks very much for taking the time to help me out; perhaps some others will benefit from this as well, as I'm sure there are more people out there who find themselves in the same position I was in. Thanks again!

JaragonCR, 2021-06-11:
@AlexProgrammerDE  @eiddor  
 In Balena 2.8.0.3+rev1 the issue is still present but the workaround is ignored because a conf file is in use.

The workaround is now 

vi /etc/systemd/system/bluetooth.service.d/bluetooth.conf

the rest holds true, thanks!

eiddor, 2021-06-11:
Thanks for the heads-up, @JaragonCR!

mlyons1340, 2021-06-23:
Hi @JaragonCR , I tried the fix but was still unable to control volume with my Samsung Galaxy 21+.  Below is the contents of /etc/systemd/system/bluetooth.service.d/bluetooth.conf.  Any ideas on what else I can try?  Any help would be greatly appreciated!

`[Unit]
Requires=bind-var-lib-bluetooth.service
After=bind-var-lib-bluetooth.service

[Service]
ExecStart=
ExecStart=/usr/libexec/bluetooth/bluetoothd --experimental --plugin=a2dp
~
~
~
~
~
~
~
~
- /etc/systemd/system/bluetooth.service.d/bluetooth.conf 1/7 14%`

Device information
Raspberry pi 4
balenaOS version: balenaOS 2.80.3+rev1

JaragonCR, 2021-06-23:
@mlyons1340 Here's my file, I just restest this with 2.80.5 +rev1 still works. What Pi are you using?

cat /etc/systemd/system/bluetooth.service.d/bluetooth.conf 
[Unit]
Requires=bind-var-lib-bluetooth.service
After=bind-var-lib-bluetooth.service

[Service]
ExecStart=
ExecStart=/usr/libexec/bluetooth/bluetoothd --experimental --plugin=a2d

Here's my systemctl output 

● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/bluetooth.service.d
             └─bluetooth.conf
     Active: active (running) since Wed 2021-06-23 20:00:09 UTC; 2h 18min ago
       Docs: man:bluetoothd(8)
   Main PID: 1738 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 2190)
     Memory: 2.4M
     CGroup: /system.slice/bluetooth.service
             └─1738 /usr/libexec/bluetooth/bluetoothd --experimental --plugin=a2dp

mlyons1340, 2021-06-23:
Thanks for the quick reply @JaragonCR!  I'm running Raspberry pi 4 with balenaOS 2.80.3+rev1.  My systemctl is below and bluetooth.conf seems to match yours exactly (i'm assuming there's a missing 'p' at the end of yours right?).

```
root@balena:~# systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/bluetooth.service.d
             └─bluetooth.conf
     Active: active (running) since Fri 2021-06-18 22:17:58 UTC; 5 days ago
       Docs: man:bluetoothd(8)
   Main PID: 1602 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 4294)
     Memory: 2.7M
     CGroup: /system.slice/bluetooth.service
             └─1602 /usr/libexec/bluetooth/bluetoothd --experimental --plugin=a2dp
```

mlyons1340, 2021-06-23:
@JaragonCR nevermind! Just needed a reboot.  Which I swear I had done lol. Thank you for your help!

JaragonCR, 2021-06-23:
> t. Which I swear I had done lol. Thank you for your help!

Glad you are up ;)

brmumm, 2021-08-21:
Hi,
I'm also having the issue of no volume control from phone when using bluetooth connection.
I'm using a Galaxy S21 phone and running OS 2.80.5 + rev1 on a Rpi 4, 4gb. with HiFiBerry Amp2.

I implemented the suggested changes and rebooted but still no gradual volume control (0% mutes, but all other volumes are 100%).  Below is a copy of my bluetooth.conf file.  Note the error messages at the end.  I'm a rookie at this so I'm sure it's an obvious fix...please forgive my ignorance.

Any and all help is greatly appreciated!

root@balena:~# cat /etc/systemd/system/bluetooth.service.d/bluetooth.conf
[Unit]
Requires=bind-var-lib-bluetooth.service
After=bind-var-lib-bluetooth.service

[Service]
ExecStart=
ExecStart=/usr/libexec/bluetooth/bluetoothd --experimental --plugin=a2dp
root@balena:~# bluetooth.service - Bluetooth service
bash: bluetooth.service: command not found
root@balena:~# systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/etc/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
Aug 21 16:10:08 balena bluetoothd[1795]: Failed to Read Adv Monitor Features with status 0x01
Aug 21 16:10:13 balena bluetoothd[1795]: ../bluez-5.58/src/service.c:btd_service_connect() a2dp-source profile connect failed for 8C:DE:E6:5B:86:42: Protocol not available
Aug 21 16:10:18 balena bluetoothd[1795]: ../bluez-5.58/src/service.c:btd_service_connect() a2dp-source profile connect failed for 8C:DE:E6:5B:86:42: Protocol not available
Aug 21 16:10:46 balena bluetoothd[1795]: ../bluez-5.58/src/service.c:btd_service_connect() a2dp-source profile connect failed for 8C:DE:E6:5B:86:42: Protocol not available
Aug 21 16:10:51 balena bluetoothd[1795]: ../bluez-5.58/src/service.c:btd_service_connect() a2dp-source profile connect failed for 8C:DE:E6:5B:86:42: Protocol not available
Aug 21 16:10:56 balena bluetoothd[1795]: ../bluez-5.58/src/service.c:btd_service_connect() a2dp-source profile connect failed for 8C:DE:E6:5B:86:42: Protocol not available
Aug 21 16:11:04 balena bluetoothd[1795]: Endpoint registered: sender=:1.217 path=/MediaEndpoint/A2DPSink/sbc
Aug 21 16:11:04 balena bluetoothd[1795]: Endpoint registered: sender=:1.217 path=/MediaEndpoint/A2DPSource/sbc
Aug 21 16:20:40 balena bluetoothd[1795]: /org/bluez/hci0/dev_8C_DE_E6_5B_86_42/fd0: fd(23) ready
Aug 21 16:31:01 balena bluetoothd[1795]: ../bluez-5.58/src/profile.c:ext_io_disconnected() Unable to get io data for Headset unit: getpeername: Transport endpoint is not connected (107)
~

JaragonCR, 2021-09-02:
Same OS I'm using, can you reboot, unpair and repair your cellphone? It's actually same OS same cellphone. HOST OS VERSION
balenaOS 2.80.5+rev1
development

brmumm, 2021-09-02:
JaragonCR --Thank you very much.  Your suggestions solved the problem.  FWIW Here's what I did (this is with a balena-sound system with the "--plugin=a2dp" already added):
1) Rebooted Rpi running balena sound
2) Unpaired balena sound device on phone
3) Rebooted phone
4) Re-paired balena sound device on phone.

I had previously rebooted the Rpi several times, but this was the first time I un-paired/re-booted/re-paired the phone.
Anyway all good...and thank you again for the fix.

tmigone, 2022-01-06:
Ok, long overdue but this is being fixed with https://github.com/balenablocks/audio/pull/87. Thanks for being patient 😮‍💨

closed 2022-02-03 by tmigone