iotsound/iotsound #303: Disabled plugins keep trying to restart, disrupting release deployments

ChestersGarage, 2020-09-23:
**Describe the problem**
Plugins that have been disabled via the environment variables DISABLE_PLUGIN_<plugin> keep trying to restart themselves and then exit, due to being disabled.  On my RPi 4B this doesn't appear to be a big deal.  But on my RPi Zero W and RPi B+, it disrupts the ability of the balenaOS supervisor to control the services.  It is causing new releases from being deployed successfully, and updates like adding or changing environment variables from taking effect in a timely manner on the slower hardware.  Once it does finally deploy, things are great.  On the faster hardware, it doesn't appear to be as impactful.  But it's overhead that is sapping up compute resources on the RPis.  The more plugins are disabled, the worse the impact.

**To Reproduce**
1. Deploy balenaSound 3.x to a RPi Zero W or RPi B+.
2. Update env vars to disable UPnP, Airplay, Spotify and/or Bluetooth.
3. Allow the updates to take effect.
4. Observe the services constantly trying to restart themselves and exiting due to being disabled.
5. Attempt to deploy a new release or even just add new env vars. (Intermittent: may need to make several attempts in order to capture a failure)
6. Observe how the disabled services take extremely long to update, relative to the rest of the services in the application.

**Expected behavior**
All services to reliably stop and restart, and in a reasonable time as needed to handle proper updates.

**Log Snippet**
22.09.20 21:06:29 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:05:28 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:06:29 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:06:30 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:07:30 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:06:29 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:07:30 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:07:31 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:08:31 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:07:30 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:08:31 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:08:32 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:09:33 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:08:31 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:09:33 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:09:33 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:10:34 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:09:33 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:10:34 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:10:34 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:10:52 (-0700)  audio  W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 335.00 ms
22.09.20 21:11:35 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:10:34 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:11:35 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:11:36 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:12:36 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:11:35 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:12:36 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:12:37 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:13:37 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:12:36 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:13:37 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:13:38 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:14:39 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:13:37 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:14:39 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:14:39 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:15:40 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:14:39 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:15:40 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:15:40 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:16:41 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:15:40 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:16:41 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:16:42 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:17:42 (-0700) Restarting service 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'
22.09.20 21:16:41 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:17:42 (-0700)  upnp  UPnP is disabled, exiting...
22.09.20 21:17:44 (-0700) Service exited 'upnp sha256:45fb1433b00d864613f161e60d2baae929a6b8cda82c546fb171f54d9e1dfa8f'

**Device information**
 - Board (hardware info): Raspberry Pi Zero W and Raspberry Pi B+
 - Device type (as seen in balenaCloud): Both reported as Raspberry Pi (v1 / Zero / Zero W)
 - balenaOS version: Both using balenaOS 2.54.2+rev1

**Additional Information**
These slower devices work great as clients, other than the impacts of the service restarts.

tmigone, 2020-09-23:
Another good catch, thanks! #304 should fix this

tmigone, 2020-09-25:
Fixed by #304.

closed 2020-09-25 by tmigone