iotsound/iotsound #305: Allo Boss DAC

johan489, 2020-09-23:
Could you pretty please add ALLO BOSS DAC to the supported HAT's.
It is an fantastic DAC.

Thanks,

Johan

tmigone, 2020-09-25:
Hey @johan489 is this the DAC? https://allo.com/sparky/boss-dac.html

johan489, 2020-09-25:
Yes, that's it.

tmigone, 2020-09-25:
Can you try following the [DAC configuration section](https://sound.balenalabs.io/docs/audio-interfaces/#dac-boards) with this value `allo-boss-dac-pcm512x-audio` for `BALENA_HOST_CONFIG_dtoverlay`? I think that's the correct one. Let me know if it works and I'll add it to the list.

johan489, 2020-09-25:
I will try it out.

tmigone, 2020-09-25:
If you already pushed the application you are going to have to re-push it because of this bug: #296. 
It should be fixed as of today but if you deployed your app yesterday then you need to re-reploy.

johan489, 2020-09-28:
`That worked perfectly, case closed

Thanks!`

closed 2020-09-28 by tmigone