iotsound/iotsound #306: No Audio Output to HiFiBerry Amp2

pbharrell, 2020-09-24:
I have been trying to troubleshoot this issue for many hours and finally resorted to asking for help.

I am running a Raspberry Pi Model B and a HiFi Berry Amp2 with the newest build of Balena OS and everything installed for Balena Sound. Long story short, I cannot get the DAC to work even though I followed the guide in this link:
[https://sound.balenalabs.io/docs/audio-interfaces/](official troubleshooting guide)

Interestingly, I get sound out of the headphone jack when the DAC is disconnected, but not otherwise. When the DAC is connected, I get nothing from the headphone jack or the DAC. I suspect it is because of these lines when initializing the audio module:
[image]

Any help you could provide would be greatly appreciated, as I am very stuck. I attached the full log in a text file below, as well. (You can skip to the very end of it to see where the error is when I restarted the audio module)
[balena-sound-1_hidden-water-23.09.20_21 13 46_(-0400).txt](https://github.com/balenalabs/balena-sound/files/5272879/balena-sound-1_hidden-water-23.09.20_21.13.46_.-0400.txt)

peteretep, 2020-09-24:
Are you using version 3, and if so, have you tried using version 2.X.X ?
Using a different DAC, I had it working fine with 2 but it doesn't work with 3.

pbharrell, 2020-09-24:
Thank you for the help! Okay, so I pushed version 2.4.2 and things are looking a little bit more promising. Now when I have the DAC connected the headphone jack still works, but no output to the Amp2. No more errors in the log, but I think it is only outputting to the headphone jack and not the DAC. Do you know how to check if the Amp2 is recognized on this version? Before I could use the audio service, but that service isn't in this version.

peteretep, 2020-09-24:
when using the version 2, you need to set a DT parameter `audio=off` - did you do that?
[image]

pbharrell, 2020-09-24:
Oh no, I didn't realize that actually. I can change that in about an hour, I am in class until then. You still have your custom configuration variable for your DAC, right?

pbharrell, 2020-09-24:
It worked! Thank you so much for the help, I really appreciate it.

closed 2020-09-24 by pbharrell