iotsound/iotsound #307: References to the balena forums in the documentation are not clear enough.

ChestersGarage, 2020-09-24:
The documentation has a few interspersed links to the support page, which itself has only one link to the forums at the very bottom of the page.  There should be a more clear reference to the forums, in order to reduce the number of resolvable issues being reported as bugs.  While this feels well-organized from the publisher perspective, as a user, it's difficult to find the forums, and I've seen people creating bug tickets on GitHub for issues they are simply having troubles with.  It took me a long time to find any links to the forums, and I already know how to get there.  :)

tmigone, 2020-09-25:
This is a great observation. I think working on an almost daily basis with the project makes this type of issues hard to notice. So glad you brought this up!

tmigone, 2020-09-25:
Closing as your PR was merged.

closed 2020-09-25 by tmigone