iotsound/iotsound #312: Unable to get DAC working on Pi Zero W

itsthekraken, 2020-09-28:
**Describe the problem**
The Pi Zero does not load the DAC correctly and therefor I have no sound output. Tried to fix the issue by following issue #296 but that did not solve the issue. Also cannot update the OS version to a newer version.

**Device information**
 - Board (hardware info): Pi zero Wireless
 - Device type (as seen in balenaCloud): Raspberry Pi (v1 / Zero / Zero W)
 - balenaOS version:  balenaOS 2.54.2+rev1

**Additional context**
Output from the device:
27.09.20 18:47:13 (+0200)  audio  Waiting for sound supervisor to start
27.09.20 18:47:13 (+0200)  audio  Setting audio routing rules. Note that this can be changed after startup.
27.09.20 18:47:13 (+0200)  audio  Routing 'balena-sound.input' to 'balena-sound.output'.
27.09.20 18:47:13 (+0200)  audio  Routing 'balena-sound.output' to 'alsa_output.dac.stereo-fallback'.
27.09.20 18:47:14 (+0200)  audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
27.09.20 18:47:18 (+0200)  audio  E: [pulseaudio] module-loopback.c: No such sink.
27.09.20 18:47:18 (+0200)  audio  E: [pulseaudio] module.c: Failed to load module "module-loopback" (argument: "source="balena-sound.output.monitor" sink="alsa_output.dac.stereo-fallback""): initialization failed.
27.09.20 18:47:18 (+0200)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
27.09.20 18:47:18 (+0200)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
27.09.20 18:47:18 (+0200)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
27.09.20 18:47:18 (+0200)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
27.09.20 18:47:18 (+0200)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11

peteretep, 2020-09-28:
what DAC are you using?

itsthekraken, 2020-09-28:
> what DAC are you using?

Sorry, its the allo Boss mini dac.

root@19eedc02d6d5:/usr/src# pactl list sinks
Sink #0
        State: SUSPENDED
        Name: alsa_output.0.stereo-fallback
        Description: Built-in Audio Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 6
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.0.stereo-fallback.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY 
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "Boss DAC HiFi [Master] pcm512x-hifi-0"
                alsa.id = "Boss DAC HiFi [Master] pcm512x-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "BossDAC"
                alsa.long_card_name = "BossDAC"
                alsa.driver_name = "snd_soc_allo_boss_dac"
                device.bus_path = "/devices/platform/soc/soc:sound/sound/card0"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card0"
                device.form_factor = "internal"
                device.string = "hw:0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.profile.name = "stereo-fallback"
                device.profile.description = "Stereo"
                device.description = "Built-in Audio Stereo"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Ports:
                analog-output: Analog Output (priority: 9900)
        Active Port: analog-output
        Formats:
                pcm

Sink #1
        State: IDLE
        Name: balena-sound.input
        Description: Null Output
        Driver: module-null-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 22
        Mute: no
        Volume: front-left: 49152 /  75% / -7.50 dB,   front-right: 49152 /  75% / -7.50 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: balena-sound.input.monitor
        Latency: 55304 usec, configured 66666 usec
        Flags: DECIBEL_VOLUME LATENCY 
        Properties:
                device.description = "Null Output"
                device.class = "abstract"
                device.icon_name = "audio-card"
        Formats:
                pcm

Sink #2
        State: RUNNING
        Name: balena-sound.output
        Description: Null Output
        Driver: module-null-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 23
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: balena-sound.output.monitor
        Latency: 54898 usec, configured 66666 usec
        Flags: DECIBEL_VOLUME LATENCY 
        Properties:
                device.description = "Null Output"
                device.class = "abstract"
                device.icon_name = "audio-card"
        Formats:
                pcm

Sink #3
        State: SUSPENDED
        Name: snapcast
        Description: Unix FIFO sink /var/cache/snapcast/snapfifo
        Driver: module-pipe-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 24
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: snapcast.monitor
        Latency: 0 usec, configured 0 usec
        Flags: DECIBEL_VOLUME LATENCY 
        Properties:
                device.string = "/var/cache/snapcast/snapfifo"
                device.description = "Unix FIFO sink /var/cache/snapcast/snapfifo"
                device.icon_name = "audio-card"
        Formats:
                pcm

tmigone, 2020-09-28:
hey @itsthekraken can you post the output of `dmesg` command here? thanks!

itsthekraken, 2020-09-28:
`root@19eedc02d6d5:/usr/src# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.118 (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Fri Aug 14 08:34:34 UTC 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1e400000
[    0.000000] On node 0 totalpages: 126976
[    0.000000]   Normal zone: 1116 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 126976 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x94/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:75:5F:3A vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 rootfstype=ext4 rootwait root=UUID=ba1eadef-d609-45c5-ba79-ed48222e8b97 rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 473468K/507904K available (8055K kernel code, 683K rwdata, 2360K rodata, 8640K init, 802K bss, 26244K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
                   lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (8056 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (8640 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 684 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 803 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 28694 entries in 85 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000092] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000222] bcm2835: system timer (irq = 27)
[    0.000648] Console: colour dummy device 80x30
[    0.001299] console [tty1] enabled
[    0.001391] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060402] pid_max: default: 32768 minimum: 301
[    0.060965] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061028] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062499] CPU: Testing write buffer coherency: ok
[    0.063911] Setting up static identity map for 0x8200 - 0x8238
[    0.065400] devtmpfs: initialized
[    0.076683] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.077179] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.077261] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.078658] pinctrl core: initialized pinctrl subsystem
[    0.080321] NET: Registered protocol family 16
[    0.083841] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.091778] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.091852] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.092018] Serial: AMBA PL011 UART driver
[    0.095358] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.110704] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-06-01 13:25, variant start_cd
[    0.120724] raspberrypi-firmware soc:firmware: Firmware hash is 6379679d1ec6a8c746d7e77e015f5b56b939976f
[    0.174493] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.177193] SCSI subsystem initialized
[    0.177526] usbcore: registered new interface driver usbfs
[    0.177689] usbcore: registered new interface driver hub
[    0.177961] usbcore: registered new device driver usb
[    0.180404] clocksource: Switched to clocksource timer
[    0.294222] VFS: Disk quotas dquot_6.6.0
[    0.294420] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.294773] FS-Cache: Loaded
[    0.295176] CacheFiles: Loaded
[    0.314973] NET: Registered protocol family 2
[    0.316340] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[    0.316468] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[... 293 lines cut ...]
[61852.485476] br-7241b1b409af: port 2(veth52f8dfb) entered disabled state
[61852.558272] device veth52f8dfb entered promiscuous mode
[61852.635930] IPv6: ADDRCONF(NETDEV_UP): veth52f8dfb: link is not ready
[61853.381871] br-7241b1b409af: port 3(veth1b12cfc) entered blocking state
[61853.390347] br-7241b1b409af: port 3(veth1b12cfc) entered disabled state
[61853.505559] device veth1b12cfc entered promiscuous mode
[... cut ...]
[61882.418022] br-7241b1b409af: port 3(veth1b12cfc) entered disabled state
[61882.581710] device veth1b12cfc left promiscuous mode
[61882.588750] br-7241b1b409af: port 3(veth1b12cfc) entered disabled state
[61884.357654] br-7241b1b409af: port 2(veth52f8dfb) entered disabled state
[61884.392855] vethd099317: renamed from eth0
[61884.572371] br-7241b1b409af: port 2(veth52f8dfb) entered disabled state
[61884.648975] device veth52f8dfb left promiscuous mode
[61884.655907] br-7241b1b409af: port 2(veth52f8dfb) entered disabled state`

itsthekraken, 2020-09-28:
root@19eedc02d6d5:/usr/src# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.118 (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Fri Aug 14 08:34:34 UTC 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1e400000
[    0.000000] On node 0 totalpages: 126976
[    0.000000]   Normal zone: 1116 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 126976 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x94/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:75:5F:3A vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 rootfstype=ext4 rootwait root=UUID=ba1eadef-d609-45c5-ba79-ed48222e8b97 rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 473468K/507904K available (8055K kernel code, 683K rwdata, 2360K rodata, 8640K init, 802K bss, 26244K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
                   lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (8056 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (8640 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 684 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 803 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 28694 entries in 85 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000092] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000222] bcm2835: system timer (irq = 27)
[    0.000648] Console: colour dummy device 80x30
[    0.001299] console [tty1] enabled
[    0.001391] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060402] pid_max: default: 32768 minimum: 301
[    0.060965] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061028] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062499] CPU: Testing write buffer coherency: ok
[    0.063911] Setting up static identity map for 0x8200 - 0x8238
[    0.065400] devtmpfs: initialized
[    0.076683] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.077179] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.077261] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.078658] pinctrl core: initialized pinctrl subsystem
[    0.080321] NET: Registered protocol family 16
[    0.083841] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.091778] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.091852] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.092018] Serial: AMBA PL011 UART driver
[    0.095358] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.110704] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-06-01 13:25, variant start_cd
[    0.120724] raspberrypi-firmware soc:firmware: Firmware hash is 6379679d1ec6a8c746d7e77e015f5b56b939976f
[    0.174493] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.177193] SCSI subsystem initialized
[    0.177526] usbcore: registered new interface driver usbfs
[    0.177689] usbcore: registered new interface driver hub
[    0.177961] usbcore: registered new device driver usb
[    0.180404] clocksource: Switched to clocksource timer
[    0.294222] VFS: Disk quotas dquot_6.6.0
[    0.294420] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.294773] FS-Cache: Loaded
[    0.295176] CacheFiles: Loaded
[    0.314973] NET: Registered protocol family 2
[    0.316340] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[    0.316468] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[... 293 lines cut ...]
[61852.485476] br-7241b1b409af: port 2(veth52f8dfb) entered disabled state
[61852.558272] device veth52f8dfb entered promiscuous mode
[61852.635930] IPv6: ADDRCONF(NETDEV_UP): veth52f8dfb: link is not ready
[61853.381871] br-7241b1b409af: port 3(veth1b12cfc) entered blocking state
[61853.390347] br-7241b1b409af: port 3(veth1b12cfc) entered disabled state
[61853.505559] device veth1b12cfc entered promiscuous mode
[... cut ...]
[61882.418022] br-7241b1b409af: port 3(veth1b12cfc) entered disabled state
[61882.581710] device veth1b12cfc left promiscuous mode
[61882.588750] br-7241b1b409af: port 3(veth1b12cfc) entered disabled state
[61884.357654] br-7241b1b409af: port 2(veth52f8dfb) entered disabled state
[61884.392855] vethd099317: renamed from eth0
[61884.572371] br-7241b1b409af: port 2(veth52f8dfb) entered disabled state
[61884.648975] device veth52f8dfb left promiscuous mode
[61884.655907] br-7241b1b409af: port 2(veth52f8dfb) entered disabled state

itsthekraken, 2020-10-05:
Hi, I created a new deployment from the git repo and that did solve the DAC issue but now the sound with spotify is choppy and not fun to listen to and airplay just doesn't start.

05.10.20 18:34:39 (+0200)  spotify  ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
05.10.20 18:34:53 (+0200)  spotify  ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
05.10.20 18:35:07 (+0200)  airplay  fatal error: could not connect to the pulseaudio playback stream -- the error message is "Bad state".

tmigone, 2020-10-06:
hey @itsthekraken are you running in `STANDALONE` mode?

itsthekraken, 2020-10-09:
> hey @itsthekraken are you running in `STANDALONE` mode?

Yes it is in standalone mode. 

I did some more testing and have rebooted the pi several times but now it does work normally.

tmigone, 2020-10-28:
Sorry for the delayed response, glad to know it's working normally now!

closed 2020-10-28 by tmigone