iotsound/iotsound #314: usb-connected hifiberry-dacplus not working after 3.x upgrade

eiddor, 2020-09-29:
**Describe the problem**
I've been running balenaSound for a long time using FiiO E10K USB DAC and the hifiberry-dacplus dtoverlay.  I upgraded to from 2.4.10 to 3.1.5 today and the (USB-connected) DAC no longer works (no sound via Bluetooth or Spotify).  If I push 2.4.10 back to the device, the DAC functions again.  I can reproduce this at will in case there are specific logs/commands that you need.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade from 2.4.10 to 3.1.5
2. Play music via Spotify or Bluetooth

**Expected behavior**
Sound from the USB-connected DAC

**Screenshots**
N/A

**Device information**
 - Board (hardware info): Raspberry Pi 3B+
 - Device type (as seen in balenaCloud): Raspberry Pi 3
 - balenaOS version: balenaOS 2.38.0+rev1

**Additional context**
Add any other context about the problem here.

tmigone, 2020-09-30:
hi @eiddor can you post the output of the following:
- `http://<DEVICE_IP>:3000/support`, where `<DEVICE_IP>` is the Pi's IP address as seen in the dashboard
- `dmesg` command run in the `audio` container. 

Thanks! You can post directly here or using a pastebin.

eiddor, 2020-09-30:
hi @tmigone - See below:

- support info - https://pastebin.com/VLhze2LC

- dmesg from audio container - https://pastebin.com/Q55cMXv0

Thanks!

tmigone, 2020-10-06:
hey @eiddor a few notes on your case. I can't spot anything wrong in the logs you provided, so let's check a couple things:

- Starting with 3.0 you no longer need to set `RESIN/BALENA_HOST_CONFIG_dtparam` to `audio=off` for DACs. Not sure if this was the case but be sure remove the config variable if you added it. 
- Again starting with 3.0 USB DAC's should (mostly) work out of the box with no configuration required. So I would try and remove the `hifiberry-dacplus` dtoverlay and see if it picks it up (from the logs I think it should). Also interested to know where you read about using that dtoverlay for the USB DAC, if you can provide more info i'm curious. 

Let me know how this goes.

eiddor, 2020-10-06:
Hi @tmigone - Let me give your dtparam suggestions a try.  I should have some results momentarily.   

I've had this thing running for a long time, so my memory is fuzzy on why I added dtoverlay settings when I'm using a USB DAC, but my guess is that I was following the early docs/comments, so I was trying different things before I finally got it working, and I just never went back to remove it.  I did try removing it while testing before opening the issue and it didn't help. 

PS - The new docs are much clearer on the differences between USB and DAC, so that's really appreciated.

eiddor, 2020-10-06:
@tmigone No luck.

I tried resetting the dtparam back to default ("i2c_arm=on","spi=on","audio=on") and also removing it entirely.

I also made sure dtoverlay wasn't defined and tried to force SOUND_OUTPUT=USB.

The only "errors" I see in the console are:

```
06.10.20 15:27:41 (-0500)  audio  E: [pulseaudio] module-loopback.c: No such sink.
06.10.20 15:27:41 (-0500)  audio  E: [pulseaudio] module.c: Failed to load module "module-loopback" (argument: "source="balena-sound.output.monitor" sink="alsa_output.Audio.analog-stereo""): initialization failed.
06.10.20 15:27:42 (-0500)  audio  E: [pulseaudio] main.c: Sink alsa_output.Audio.analog-stereo does not exist.
06.10.20 15:27:42 (-0500)  audio  E: [pulseaudio] main.c: Sink alsa_output.Audio.analog-stereo does not exist.
06.10.20 15:27:42 (-0500)  audio  E: [pulseaudio] main.c: Sink alsa_output.Audio.analog-stereo does not exist.
06.10.20 15:27:42 (-0500)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
06.10.20 15:27:42 (-0500)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
```

Pushing 2.4.10 back to the Rpi makes everything work again with the same settings/parameters (after setting audio=off).

julianatanasoae, 2020-10-10:
I'm having the exact same issue with a Focusrite USB Scarlett Solo. It worked before in 2.x.

cpwood, 2020-10-23:
FWIW, I was able to workaround this temporarily by getting an SSH session in the audio service and issuing the following commands:

`pactl list`

to find the name of the sink for my USB DAC, and then using that name:

`pactl set-default-sink alsa_output.usb-soundcard-1.analog-stereo`

I ended up cloning the repo and changing `core\audio\start.sh` as follows:

```bash
# Route "balena-sound.output" to the appropriate audio hardware
function route_output_sink() {
  #local OUTPUT=""

  # Audio block outputs the default sink name to this file
  # If file doesn't exist, default to sink #0. This shouldn't happen though
  #local SINK_FILE=/run/pulse/pulseaudio.sink
  #if [[ -f "$SINK_FILE" ]]; then
  #  OUTPUT=$(cat "$SINK_FILE")
  #fi
  #OUTPUT="${OUTPUT:-0}"

  # WooAudio customisation
  local OUTPUT="alsa_output.usb-soundcard-1.analog-stereo"
  
  sed -i "s/%OUTPUT_SINK%/sink=\"$OUTPUT\"/" "$CONFIG_FILE"
  echo "Routing 'balena-sound.output' to '$OUTPUT'."
}
```

eiddor, 2020-10-23:
Thanks @cpwood - This worked perfectly for me!

eiddor, 2020-10-23:
Ok, I've been fumbling around in the code and the audio container and I _think_ that I see the problem, but I'm not skilled or confident enough in source code (or the mechanics of how we're doing audio) to fix it properly.

The offending device is coming from here - 

```
root@44019a741b7f:~# cat /run/pulse/pulseaudio.sink 
alsa_output.Audio.analog-stereo
```
I did try to dig around in the Balena audio architecture docs for a clue, but it's well above my skill level.

My bet is that @tmigone will see the problem right away :-)

eiddor, 2020-10-24:
So after some further digging (this might be obvious to everyone, so I'm sorry if I'm stuck on the basics), it looks like the following line in: 

https://github.com/balenablocks/audio/blob/master/entry.sh is determining the default sink for PulseAudio.

`USB_CARD=$(cat /proc/asound/cards | awk -F'\[|\]:' '/usb|USB|Usb/ && NR%2==1 {gsub(/ /, "", $0); print $2}' | awk 'NR==1')`

The result is "Audio" which then gets formatted to "`alsa_output.Audio.analog-stereo`" later in the script.

In my case I need it to be `alsa_output.usb-soundcard-0.analog-stereo`

tmigone, 2020-10-28:
Thanks @eiddor that's some serious investigative work and definitely saved me a bunch of time 😄 🥳 

As you suggest, the problem here is the audio script is not properly recognising/tagging the usb sink. I'll see if I can replicate this with some of my audio gear, though as I haven't ran into this problem yet I suspect I wont be able to. 

If someone is comfortable sharing access with me to their device to run a few quick tests that would help a lot, let me know and I'll explain how to do that safely.

**Workarounds**
In the meantime, here are some workarounds gathered from this thread (thanks @cpwood and @eiddor):

First we need to figure out what the USB sound card name is. This can be done by running the following command in the `audio` service: 

```
pactl list sinks short
```

We are looking for a sink that's most likely named `alsa_output.usb-soundcard-<ID>.analog-stereo`.

Once you figure out what the sink name is there are a few options:
- Get into the `audio` service and run `pactl set-default-sink alsa_output.usb-soundcard-<ID>.analog-stereo`. Note that this changes will be lost if you restart the device/service. 
- Fork the repo and force the default sink to the correct value as @cpwood did [here](https://github.com/balenalabs/balena-sound/issues/314#issuecomment-715248194).
- Set the `SOUND_OUTPUT/AUDIO_OUTPUT` env var to the correct sink value: `SOUND_OUTPUT=alsa_output.usb-soundcard-<ID>.analog-stereo`. This can be done via balenaCloud's dashboard. **Note**: `SOUND_OUTPUT` was "deprecated" in favour of "AUDIO_OUTPUT", it was not working anyways due to a bug.

eiddor, 2020-10-28:
@tmigone You can access mine if you want to ping me and let me know how to enable that.  I'm currently upgrading back to 3.19 so you can do your thing.  (I also reverted to 2.x to get around the bluetooth volume control issue).

Just FYI, the balena sound doc site isn't working and I'm getting this on any page under sound.balenalabs.io that I try to visit.

[image]

tmigone, 2020-10-29:
@eiddor thanks again man, I've identified the issue already using your device to run some tests, appreciate it. I'll get the fix out in a few moments.

About our docs site, yeah. We have some sort of problem with the bot that produces the documentation site on every merge to master. I'm trying to chase down what the issue is. Hopefully we can get it back online soon.

eiddor, 2020-10-29:
Thanks, @tmigone - That's excellent news!!  I really appreciate the help!

tmigone, 2020-10-29:
All right, this should be now fixed! 
cc: @eiddor @julianatanasoae @cpwood 

To get the fix you'll need to re-deploy balenaSound. You need to do it even if you are already running the latest version (v3.1.11 at the time I'm writing this). This is because the fix was not made on this repository but rather on the [audio block](https://github.com/balenablocks/audio/commit/b82052f7ba6b167da44f00088b84717e2d682269) repo which this project depends on.

If anyone is interested on the fix you can find it [here](https://github.com/balenablocks/audio/commit/b82052f7ba6b167da44f00088b84717e2d682269), the problem and fix were exactly as described by @eiddor [here](https://github.com/balenalabs/balena-sound/issues/314#issuecomment-716040512). 

I'll leave this open until someone confirms it works, cheers!

eiddor, 2020-10-29:
All good here, @tmigone - This is from my console:

```
29.10.20 15:03:40 (-0500)  audio  Setting audio routing rules. Note that this can be changed after startup.
29.10.20 15:03:40 (-0500)  audio  Routing 'balena-sound.input' to 'balena-sound.output'.
29.10.20 15:03:40 (-0500)  audio  Routing 'balena-sound.output' to 'alsa_output.usb-soundcard-0.analog-stereo'.
```

Thank you again!

tmigone, 2020-10-29:
Awesome! Closing the issue then, if someone else runs into trouble let me know.

closed 2020-10-29 by tmigone

Zageron, 2021-09-09:
@tmigone `pactl list sinks short` is no longer displaying my USB DACs.
```sh
bash-5.0# pactl list sinks short
0       alsa_output.bcm2835-jack.stereo-fallback        module-alsa-card.c      s16le 2ch 44100Hz     RUNNING
1       balena-sound.input      module-null-sink.c      s16le 2ch 44100Hz       IDLE
2       balena-sound.output     module-null-sink.c      s16le 2ch 44100Hz       IDLE
3       snapcast        module-null-sink.c      s16le 2ch 44100Hz       RUNNING
```
I've tried two Schiit Modi 2.

Same as:
#493