iotsound/iotsound #316: SOUND_DEVICE_NAME isn't affecting the Bluetooth device name

ChestersGarage, 2020-10-01:
**Describe the problem**
The value set for SOUND_DEVICE_NAME is not being used as the device's Bluetooth name.  Despite setting my device's SOUND_DEVICE_NAME="Multi-Room Audio ZW", the device appears in my phone's Bluetooth screen as "balenaOS e714"

**To Reproduce**
Steps to reproduce the behavior:
1. Set SOUND_DEVICE_NAME=<my_device_name> on a device
2. Allow the device to update and settle.
3. Search for the device in your phone's Bluetooth utility.
4. Note the BT name is not what was set in the env var.

**Expected behavior**
The device should show up in BT with the name set by SOUND_DEVICE_NAME.

**Device information**
 - Board (hardware info): Raspberry Pi 4B
 - Device type (as seen in balenaCloud): Raspberry Pi 4
 - balenaOS version: balenaOS 2.56.0+rev1

magicalyu, 2020-10-16:
Same issue

EuanKerr, 2020-10-19:
you can use `BLUETOOTH_DEVICE_NAME` as a temporary fix until BalenaSound pulls through `SOUND_DEVICE_NAME ` properly

tmigone, 2020-10-28:
This should be addressed by #338, and I say addressed and not fixed because we'll have to use `BLUETOOTH_DEVICE_NAME` for the time being. If you are interested you can find a more detailed explanation of what I mean in the PR, also suggestions welcome if you can think of a better solution :)

closed 2020-10-28 by tmigone