iotsound/iotsound #317: Support for the Hifiberry Digi+ Pro 'DAC'

SerialHacker, 2020-10-01:
Currently, only the Digi+ and not the Digi+ Pro is included in the list of supported DACs, Hifiberry recommended that i use the dtoverlay value 'hifiberry-digi-pro' which seems to be working to a great success.

I was wondering whether the Digi+ Pro is completely supported already? Or whether further coding should be added to fully support it

Also the SOUND_VOLUME environmental variable doesn't seem to be working with 'audio=off' and 'Hifiberry-digi-pro'

shawaj, 2020-12-27:
@SerialHacker yes we already have the overlay available in the balenaOS https://github.com/balena-os/balena-raspberrypi/blob/master/layers/meta-balena-raspberrypi/conf/layer.conf

So can you confirm the HiFiBerry Digi+ Pro is working nicely for you with the `hifiberry-digi-pro` overlay?

tmigone, 2021-04-20:
Closing as the Digi+ Pro is now categorized as untested. See https://github.com/balenalabs/balena-sound/issues/439 and https://sound.balenalabs.io/docs/audio-interfaces#dac-boards

closed 2021-04-20 by tmigone