iotsound/iotsound #319: Enable  control of mutli-room devices

ejannett, 2020-10-02:
Hi,

Multiroom feature works great but is missing the ability to control volume on devices.
It is interesting sometime to be able to mute some devices or to no have the same sound volume  according to 
the room where the device lives. 

snapcast comes with a simple web interface : snapweb

Just tried, it works with minimum changes.
-1 clone https://github.com/badaix/snapweb.git as a submodule within core/multiroom/server/
-2 applied following changes
```
% git diff        
diff --git a/core/multiroom/server/Dockerfile.template b/core/multiroom/server/Dockerfile.template
index 9a6db0a..2768ed8 100644
--- a/core/multiroom/server/Dockerfile.template
+++ b/core/multiroom/server/Dockerfile.template
@@ -5,5 +5,6 @@ WORKDIR /usr/src
 RUN install_packages snapcast-server
 COPY snapserver.conf /etc/snapserver.conf
 COPY start.sh .
+COPY snapweb/dist /var/www
 
 CMD [ "/bin/bash", "/usr/src/start.sh" ]
\ No newline at end of file
diff --git a/core/multiroom/server/snapserver.conf b/core/multiroom/server/snapserver.conf
index ed1f90f..a2b0cdc 100644
--- a/core/multiroom/server/snapserver.conf
+++ b/core/multiroom/server/snapserver.conf
@@ -1,3 +1,9 @@
+[http]
+enabled = true
+bind_to_address = 0.0.0.0
+port = 1780
+doc_root = /var/www/
+
 [stream]
 stream = pipe:///var/cache/snapcast/snapfifo?name=balenaSound
 sampleformat = 44100:16:2
\ No newline at end of file
diff --git a/docker-compose.yml b/docker-compose.yml
index c26de5d..2aee346 100644
--- a/docker-compose.yml
+++ b/docker-compose.yml
@@ -31,6 +31,7 @@ services:
     ports:
       - 1704:1704
       - 1705:1705
+      - 1780:1780
     volumes:
       - snapcast:/var/cache/snapcast
 
```

Would be good feature to have this by default

ejannett, 2020-10-03:
In order to have persisted client name, volume and "muted" flag. the server config must be placed 
on persistent storage. 
following should be added to  snapserver.conf 
[server]
datadir = /var/cache/snapcast/

ChestersGarage, 2020-10-29:
@ejannett Have you looked at installing a stand-alone Snapcast client?  I run Snap.Net on my Windows PC and the Snapcast client (from badaix) on my Android phone, and both work well to manage individual device volume and mute.  They also allow some arbitrary group and device naming.  Until the web UI is built within balenaSound, the stand-alone Snapcast clients could be a superb workaround.

ejannett, 2020-10-30:
Hi,

> @ejannett Have you looked at installing a stand-alone Snapcast client? I run Snap.Net on my Windows PC and the Snapcast client (from badaix) on my Android phone, and both work well to manage [...]

Yes , I've first looked at existing client but I use iPhone and using snap web was just a matter of enabling existing things, i.e without any kind of extra setup.

ChestersGarage, 2020-10-30:
I'm looking forward to this change.  Nice addition!

ejannett, 2020-11-04:
enhancement delivered

closed 2020-11-04 by ejannett