iotsound/iotsound #323: Bluetooth as audio out

shqipesson, 2020-10-06:
Is it possible to use The onboard bluetooth or an bluetooth dongle to tu use an existing bluetooth speaker as audio out instead of The analog out? Havr tried to get it work with raspbian and raspotify with no success.. 

Have a bunch of bluetooth speakers and make them to cast devices would seem very useful for many i assume. 
 
Have a rpi 2b and a 3b i would like to implement that on..

Thanks for your replies

febs, 2020-10-09:
I'd love to have this feature as well.

magicalyu, 2020-10-15:
great feature!

swinterroth, 2021-12-22:
I'm interested in this!

chaosloth, 2022-01-22:
Ditto - same use case as OP.   I have  a ton of BT speakers littered throughout the house and would love to have them as an output, even if they are a little out of sync that would be ok as you cannot hear the speakers from one room to the next

spuksic, 2022-01-23:
This would solve all my problems.

andrewn, 2022-07-10:
I found that `balena-sound` automatically pairs with my bluetooth speaker when I turn it on.

I managed to get bluetooth audio output working by:

1. Turning on the BT speaker and waiting for `balena-sound` to automatically pair
2. Visiting `http://192.168.21.108/audio/sinks` (or whatever the local IP address is)
3. Finding the bluetooth audio sink. Mine was called `"bluez_sink.00_02_3C_6B_63_48.a2dp_sink"`
4. Add a Device Variable `PULSE_SINK` with the value of the sink name `bluez_sink.00_02_3C_6B_63_48.a2dp_sink`5. 
5. Add a Device Variable `SOUND_MODE` with value `STANDALONE`

After that, Spotify and Airplay stream to the BT speaker.

I was getting errors with the multiroom service, hence switching to the `STANDALONE` mode.

ijv182, 2022-07-12:
> I found that `balena-sound` automatically pairs with my bluetooth speaker when I turn it on.
>
> I managed to get bluetooth audio output working by:
>
> 1. Turning on the BT speaker and waiting for [...]

I'm so stoked to have found your solution after spending a better part of yesterday trying to do this through standalone Shairport-sync. 
How exactly do you go about pairing your speaker to balena-sound? I tried to use the terminal on the balenacloud page using bluetoothctl for that initial pairing but I came up short. any tips for that first step?

wheredidthecostgo, 2022-11-09:
Thanks for the tips!  I had to use a combination since my device would not auto pair

1. Turn on the Bluetooth speaker and enter discovery mode
2. Open up the command line terminal on the Raspberry Pi and run bluetoothctl
3. Power on the Bluetooth device: power on
4. Turn the agent on: agent on
5. Scan for devices: scan on
6. Pair with your Bluetooth speaker pair MAC Address, at this point there should be no passcode to enter
7. Add the new Bluetooth speaker to the list of trusted devices: trust MAC Address
8. Now finally connect to the Bluetooth speaker: connect MAC Address
9. Visiting http://192.168.21.108/audio/sinks (or whatever the local IP address is)
10. Finding the bluetooth audio sink. Mine was called "bluez_sink.00_02_3C_6B_63_48.a2dp_sink"
11. Add a Device Variable PULSE_SINK with the value of the sink name bluez_sink.00_02_3C_6B_63_48.a2dp_sink5.

kiwizznz, 2022-12-24:
Is this running from the host os? I get this error when connecting to the Trek Max speaker:

[bluetooth]# connect 10:B7:F6:03:41:92
Attempting to connect to 10:B7:F6:03:41:92
Failed to connect: org.bluez.Error.Faile

kiwizznz, 2022-12-24:
Also trying that sinks url I get a "{"error":"Unexpected token u in JSON at position 0"}" error