iotsound/iotsound #325: Doesn't compile for x86

dewitpj, 2020-10-10:
**Describe the problem**
I can't deploy to x86

**To Reproduce**
Steps to reproduce the behavior:
1. Try and deploy to x86

**Expected behavior**
All images to compile

**Screenshots**
[image]

**Device information**
 - Board (hardware info): Apple Mac Mini
 - Device type (as seen in balenaCloud): Generic x86_64
 - balenaOS version: balenaOS 2.54.2+rev1

**Additional context**
Add any other context about the problem here.

dewitpj, 2020-10-10:
Forgot to add this one:

[image]

tmigone, 2020-10-28:
Hi @dewitpj, balenaSound currently doesn't support `Generic x86_64` device type, the only x86 device with support is the Intel NUCs. That said, running balenaSound on a mac mini sounds cool, I'll try to build generic x86 images in the coming days if you are up for doing a bit of testing.

dewitpj, 2020-11-01:
Totally up for testing it 👍

tmigone, 2020-11-27:
Hey @dewitpj the images should now be available for testing. Let me know how it goes!

dewitpj, 2020-11-27:
*shakes fist @ spotify*

[Error]             Some services failed to build:
[Error]               Service: spotify
[Error]                 Error: The command '/bin/sh -c curl -sSL https://dtcooper.github.io/raspotify/key.asc | apt-key add -v - 2> /dev/null   && echo 'deb https://dtcooper.github.io/raspotify jessie main' | tee /etc/apt/sources.list.d/raspotify.list   && install_packages raspotify' returned a non-zero code: 100
[Info]              Built on x64_01
[Error]             Not deploying release.

tmigone, 2020-11-27:
Ah right, `raspotify` does not have an x86 build. I'll have to replace that with `librespot` which i think does. In the meantime you can comment out the spotify part from the `docker-compose.yml` file to see if it builds fine.

dewitpj, 2020-11-27:
I'll give that a go and see - give me a few mins :)

dewitpj, 2020-11-27:
> Ah right, `raspotify` does not have an x86 build. I'll have to replace that with `librespot` which i think does. In the meantime you can comment out the spotify part from the `docker-compose.yml` file to see if it builds fine.

Yip - that deploys fine - I see all the services running, but AirPlay doesn't show up....not sure it will ?

Keeps saying 28.11.20 08:05:08 (+1300)  airplay  Waiting for audioblock to start...

dewitpj, 2020-11-27:
...and then "audio" restarts - it does "see" the card thou

dewitpj, 2020-11-28:
Hey @tmigone,

I'm trying to get it to compile but for some reason it keeps defaulting to ARM v7:

[spotify]            ---> 6a33bba29141
[spotify]           Step 2/6 : RUN install_packages librespot
[spotify]            ---> Running in 804c5fe76ceb
[spotify]           Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/):
[spotify]           Architecture: ARM v7
[spotify]           OS: Debian Buster
[spotify]           Variant: run variant
[spotify]           Default variable(s): UDEV=off
[spotify]           Extra features:
[spotify]           - Easy way to install packages with `install_packages <package-name>` command
[spotify]           - Run anywhere with cross-build feature  (for ARM only)
[spotify]           - Keep the container idling with `balena-idle` command
[spotify]           - Show base image details with `balena-info` command

Any tips ?