iotsound/iotsound #327: Allo Piano 2.1 DAC Support

discoferret, 2020-10-14:
Hi, I am currently trying to build a multi room system using new and old RPi's. I am having issues with my Allo Piano 2,1 DAC. Please see more details @ https://forums.the balena.io/t/about-the-balenasound-category/180335
I have set the "BALENA_HOST_CONFIG_dtoverlay" to "dtoverlay=allo-piano-dac-plus-pcm512x-audio " as recommended, but I am still unable to get any audio.  

**Device information**
 - Board (hardware info): [e.g. Raspberry Pi 4B 4GB
 - Device type (as seen in balenaCloud): [e.g. Raspberry Pi 4 (using 64bit OS)]
 - balenaOS version: [e.g. balenaOS v2.56.0+rev1]

Any help or advice is greatly welcome.

tmigone, 2020-10-28:
Hello @discoferret, can you share the following information to assist you?
- output of `http://<DEVICE_IP>:3000/support` where `<DEVICE_IP>` is the IP address of your device
- Screenshot of your device's `Device configuration` tab on balenaCloud dashboard

Thanks!

shawaj, 2020-12-27:
@discoferret we don't currently have the device tree overlay for this DAC in balenaOS.

I've added a pull request at the following link to try and add it in for the future... https://github.com/balena-os/balena-raspberrypi/pull/585

shawaj, 2020-12-27:
@discoferret by the way, you can still add this manually. To do this you can download the device tree overlay here https://github.com/raspberrypi/firmware/blob/master/boot/overlays/allo-piano-dac-plus-pcm512x-audio.dtbo

You can then add it to the resin-boot/overlays folder on the SD card with balenaOS on it.

Then the device tree overlay should apply correctly

tmigone, 2021-04-20:
Closing, drivers for this DAC are now part of balenaOS though they remain untested for balenaSound. See https://github.com/balenalabs/balena-sound/issues/439

closed 2021-04-20 by tmigone