iotsound/iotsound #331: WM8960_Audio_HAT DAC board support

simonvanhove, 2020-10-22:
Hi,

is it possible to provide support for WM8960_Audio_HAT from Waveshare?
Here is their manual.

https://www.waveshare.com/w/upload/5/54/WM8960_Audio_HAT_User_Manual_EN.pdf

Greets,

Simon

shawaj, 2020-12-27:
Duplicate of #264

AlexProgrammerDE, 2020-12-29:
@shawaj may you tell me how to set a issue to a duplicate? If so please let me know.

closed 2020-12-29 by AlexProgrammerDE

shawaj, 2020-12-30:
@AlexProgrammerDE hmm I am not actually sure how you do it. All I did was put a comment saying `Duplicate of ...` and it seemed to automatically set it like that?

It seems to be a standard feature of GitHub according to this doc - https://docs.github.com/en/free-pro-team@latest/github/managing-your-work-on-github/about-duplicate-issues-and-pull-requests

shawaj, 2021-01-07:
Getting closer https://github.com/balenalabs/balena-sound/issues/264#issuecomment-756428272