iotsound/iotsound #332: Non-unique Docker MAC addresses interfering with Snapcast client/server relationship

ChestersGarage, 2020-10-23:
**Describe the problem**
Snapcast clients are registering themselves with the Snapcast server using the MAC address of the Docker container they are running in, rather than the MAC address of the device they are running on.  This is confusing the Snapcast server and causing audio streams to the client devices to conflict with each other.  And it's preventing proper client and stream management via Snapcast clients.  All clients that report the same MAC address to the Snapcast server appear to be the same client when viewed through the Snapcast volume control.  I am unable to change volume or mute one device, as the change affects all devices on that MAC address. 

**To Reproduce**
1. On a fleet with multiple balenaSound devices...
2. Use a separate Snapcast client, such as the Android Snapcast client or the Windows Snap.net client to adjust volumes. 
3. The Snapcast client may show fewer devices than are actually running.
4. Adjusting a volume slider affects more than one client.
5. This issue is very intermittent and variable, depending on factors I have not yet identified.

**Expected behavior**
Each balenaSound device shows up as its own entity in the Snapcast client, and each volume slider works as expected for each device.

**Device information**
 - Board (hardware info): Raspberry Pi 4B
 - Device type (as seen in balenaCloud): Raspberry Pi 4
 - balenaOS version: balenaOS 2.56.0+rev1

**Additional context**
I believe this is adversely affecting latency as well.  But I haven't been able to verify it.

Since the Snapserver needs to know the MAC address of the client, I'm not sure how to address this, except maybe if the hardware MAC address can be determined and provided to the Snapcast server when a client registers.

This does not appear to be a problem in the 2.4 code, only 3.1, as far as I have been able to diagnose.

thepont, 2020-10-28:
This could be fixed by starting snapclient with `--hostID $BALENA_DEVICE_UUID` in the start script for snapclient.

ChestersGarage, 2020-10-29:
Thanks @thepont for that info.  I'll try it and see how things change.

tmigone, 2020-10-29:
Nice work! Did the inclusion of `hostID` fix this?

ChestersGarage, 2020-10-29:
It fixed it enough to clear up the functional issue.  There is still a very minor visual issue around that in third-party Snapcast clients.   But that looks like a Snapcast issue, not a balenaSound issue.  I created a PR for the change.

thepont, 2020-10-29:
I like that you allowed the friendly name to be used as well, Makes it heaps more usable. 

I have been using HomeAssitant to control my devices and from that perspective it's going to force me to reconfigure when it goes live. I just wonder how that effects semver, since it changes how the clients identify with the outside world.

thepont, 2020-10-29:
Currently though, I guess the id is suspect to change on any upstream change that effects how the MAC of the container is calculated, this brings heaps more stability.

tmigone, 2020-10-30:
Sounds good, I had trouble replicating the issue though since you were saying it's very intermittent I guess I didn't spend enough time trying 😝 I'll merge the PR

tmigone, 2021-02-09:
@ChestersGarage this can be closed correct? Or at least there is not much we could improve I think.

ChestersGarage, 2021-02-09:
Ya, I had to check.  But this fix addressed the issue as far as it can, and resolved the functional problem.

closed 2021-02-09 by ChestersGarage