iotsound/iotsound #333: Balena Sound - high latency and lags

benjibutters, 2020-10-25:
Hi, I just bought a Raspberry Pi 4b to play music on my Geneva XL with AirPlay. I used this manual (https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/)  and everything worked well: on the dashboard, every service is on "running" and the device is "online". 

However, somehow the latency is extremely high (around 20sec) - for adjusting the volume as well as pausing/playing music. The issue is the same when using airplay or when using the Bluetooth. 

Enclosed is the log file. 
[balena-sound_delicate-frog-25.10.20_10_59_30_(+0100).txt](https://github.com/balenalabs/balena-sound/files/5434445/balena-sound_delicate-frog-25.10.20_10_59_30_.%2B0100.txt)

HOST OS VERSION
balenaOS 2.58.6+rev1

I have no experience in coding so it would be great if I could get some help on how to fix this..

Thanks in advance!

eng-tsmith, 2020-10-25:
I am experiencing the same issue. Multi Room streaming works fine, however for any pausing / changing volume etc. it takes around 10 sec until I can hear the changes.

maciekpoplawski, 2020-11-02:
the same for me. Extreme lag. Around 14 secs.

Theo121415, 2020-11-23:
Same issue! i get lag for everything around 18 sec

tmigone, 2021-02-09:
Hi this seems a duplicate of #294 which has been fixed with improvements made in v3.4.0 (released 27th Nov 2020). Please see attached issue for reference and report any persisting problems over there. Thanks

closed 2021-02-09 by tmigone