iotsound/iotsound #335: No Connection Notifiaction Sound is Played

m0ejo, 2020-10-26:
**Describe the problem**
I upgraded from an older version of a release from back in July (Sorry. No idea what version that was) A lot of little bugs disappeared.  However, I noticed that the "Door Bell" also disappeared along with it when i connect via Bluetooth. I have not tried any of the other services before so I'm not sure if it was or is there for those.   Upgrade is probably not the right word.  Replace is probably a better description.  I setup a whole new application, re-imaged the raspberry pi3 with the latest version of BalenaOS (balenaOS 2.58.3+rev1)  Supervisor Version is at 11.14.0 and  I assume the release version of Balena Sound is at 3.1.8  I added a RPI4 to the application as well.  The Pi3 is 32bit and the Pi4 is 64bit.  I have SOUND_VOLUME set to 100.

Referring to this link:
https://forums.balena.io/t/balena-sound-disable-start-up-and-shut-down-sound-notification/144379/3
I applied CONNECTION_NOTIFY_VOLUME =100 as both the device variable and a Bluetooth service variable globally.  It does show up when I connect to the Bluetooth service terminal and run env | sort on both devices.  I also have SOUND_MODE set to STANDALONE. 

Any Ideas?

tmigone, 2020-10-28:
Hi @m0ejo, you are right, the connection notification sound was lost with the transition to v3.0. Going from v2.x to v3 was a huge update and we missed a few small details such as this one.

Bluetooth internals are now abstracted into another repo, there is an open issue about the lack of notification sounds [here](https://github.com/balenablocks/bluetooth/issues/2). You can follow either GH issue to be alerted when this is fixed.

m0ejo, 2020-10-29:
Much thanks for the info!