iotsound/iotsound #341: SOUND_DEVICE_NAME doesn't change device name

jgottlander, 2020-10-30:
**Describe the problem**
Setting SOUND_DEVICE_NAME to custom variable doesn't change device name.
BLUETOOTH_DEVICE_NAME works as intended.

**To Reproduce**
Steps to reproduce the behavior:
1. Set SOUND_DEVICE_NAME to custom variable.
2. Spotify Connect and Airplay speaker still has default name, in my case "balenaSound Spotify/Airplay 2d8c"

**Expected behavior**
Setting SOUND_DEVICE_NAME to custom variable, should change Spotify and Airplay speaker to custom variable.

**Screenshots**
https://ibb.co/tC6Sd9F

**Device information**
 - Board (hardware info): Raspberry Pi 3B
 - Device type (as seen in balenaCloud): Raspberry Pi 3
 - balenaOS version: balenaOS 2.58.3+rev1
 - DAC: HifiBerry DAC+

eiddor, 2020-10-30:
Interesting - SOUND_DEVICE_NAME does change the Spotify Connect name for me, but not the Bluetooth Name.

[image]

[image]

jgottlander, 2020-11-04:
It worked perfectly before I upgraded to balenaOS 2.58, it was 2.53/2.54 if I remember correctly.
I've now upgraded to 2.60.1+rev1 but the problem persists.

TwoTenPvP, 2020-11-06:
> Interesting - SOUND_DEVICE_NAME does change the Spotify Connect name for me, but not the Bluetooth Name.
> 
> [image]
> 
> [image]

Bluetooth name has it's own property.  ``BLUETOOTH_DEVICE_NAME``

AlexProgrammerDE, 2020-12-29:
As far as I remember spotify has in 3.0 now its own property. You may find the info you need in the docs.

closed 2020-12-29 by AlexProgrammerDE

jgottlander, 2021-01-13:
I checked raspotify's documentation, and they use DEVICE_NAME as variable name, and I tried using only DEVICE_NAME, and it indeed changes the Spotify device name. So there seems to be a need for three different variables to cover all custom names for soundbalena.

vordenken, 2021-08-13:
I can confirm this, DEVICE_NAME changes the Spotify Connect as well as the AirPlay name for me, SOUND_DEVICE_NAME does nothing. This should probably be changed in the documentation.