iotsound/iotsound #342: UPnP music control is unstoppable

GuillaumeDen, 2020-11-02:
**Desciption**
I succesfully installed balenaSound pretty quickly.
The bluetooth connection and playing is functional (even if there is a big delay).
Airplay is doing a great job and works like a charm.

But I am experiencing a very painful problem using UPnP : the music doesn't stop.

On my android phone (using BubbleUpnp or XCast produces the same problem) I start playing a music, and if I stop it or disconnect, the music restart and nothing is connected to my raspberry, so I can't even pause it. The only solution is to reboot/restart the device (or lower his volume but it's secretly playing 😅)

Do I have something special to do to correct this ?

**Device information**
 - Board : Raspberry Pi 3B+
 - Device type : Raspberry Pi 3
 - balenaOS version: 2.58.3+rev1

**Additional infos**
[image]
The first line appeared once I clicked play, the second once I disconnected or clicked on stop.

JodyGoldberg, 2021-01-05:
I've had a significant improvement in upnp stability by updating gmediarender from the original 0.0.7 in buster to https://github.com/hzeller/gmrender-resurrect which is packaged in ubuntu focal.    It also appears to break spotify so I won't send a patch until that's resolved.

ytzeng1, 2021-03-05:
Any idea how to patch this on our own? I am not using spotify plugin anyway, thanks!

drew-kun, 2021-03-30:
bump on this.. I would also love to test the patched gmreder, because I do not use spotify and I need to use upnp

drew-kun, 2021-03-31:
Ok, so using the **gmrender 0.0.9** from **debian-bullseye** helps - the tracks become stoppable so it is cool, but the packets are still being lost from time to time.. I thought that maybe it is because my android device is on different VLAN, but I switched to the same vlan where my balena-sound is and it kinda helped, but just a little, so the sound played over upnp still sticks sometime. I hope this could be fixed in the future, maybe when the project officially switches to **gmrender-resurrect** hopefully...