iotsound/iotsound #344: Balena sound doesn't connect to home wifi

febs, 2020-11-07:
**Describe the problem**
Flashed the image with the correct wifi parameters, booted, it won't connect to wifi.

If connected to a monitor, a log line on the top of the screen reads "wlan0: link is not ready"

**Device information**
 - Board (hardware info):  Raspberry Pi 3B+
 - Device type: Raspberry Pi 3
 - balenaOS version: 2.60.1+rev1-dev-v11.14.0

shawaj, 2020-12-26:
@febs it sounds like it could be an issue with the Raspberry Pi itself. However let's just run through a few things.

Do you have another Pi you could try it with?

Could you attach the SD card to your computer and check in the resin-boot/system-connections folder if there are any files? And if so, is there one that contains your WiFi details? 

Don't paste the your SSID and password here as it could be a security risk for you, but could you post the rest of the text from the file?

febs, 2020-12-28:
Thanks for that. I am afraid I moved to a manual snapcast setup, so I forgot which raspberry I used back then, and I don't have a spare SD card anymore. Since it looks like that was related only to myself, I am going to close this. Thanks for the feedback.

closed 2020-12-28 by febs

shawaj, 2020-12-28:
@febs great. Thanks very much for the update 😊