iotsound/iotsound #346: Airplay sound its too low

SigmaReloaded, 2020-11-09:
hi guys! I'm already using Raspberry4 4gb with balena sound, when i use airplay from my Iphone to reproduce music from spotify, slace volumen it's not working very well and output volumen its too low. I'm trying with this configs

Device Variables config
[image]

[image]

Any suggestions?

parallelbgls, 2020-11-29:
Seems to have same problem with RP4, I have to turn all devices to maximum to get suitable volumn. I think max volumn is too low.

tmigone, 2022-04-05:
Hey can you try reproduce this with latest balenaSound version 3.8.9? Feel free to re-open this issue if the problem persists. Thanks!

closed 2022-04-05 by tmigone