iotsound/iotsound #347: Device names in Snapweb not that helpful

sceptiq, 2020-11-11:
Love the Snapweb!

But would be awesome to have the 

```SOUND_DEVICE_NAME```

And not an id(Bluetooth?)

But still thx for the webinterface :)

ejannett, 2020-11-12:
Hi,

What you see in the interface is the name of snapcast client. This name (according to recent push) should be
the device ID. Snapweb allow you to alias the name . In the interface select the edit icon and choose your name 
and group. 
This name should be persisted so you have to it only once.

hope this helps

EJ

sceptiq, 2020-11-12:
Oh didn't see that - it helps - thx :)

But using the *_DEVICE_NAME would be... awesome ;)

ejannett, 2020-11-12:
Hi,

Note the name shown in the web interface is from the ID of the snapcast client which have no knowledge of *_DEVICE_NAME
EJ

tmigone, 2021-02-09:
Hi as @ejannett explained this is outside of our control as there is currently no way of specifying the snapclient name. There is a recently opened issue on SnapCast's repo (https://github.com/badaix/snapcast/issues/759) so maybe it'll gain traction and be implemented soon. Once that's done we can add it to balenaSound.

badaix, 2021-02-19:
Default name is the hostname and you can change the name using the [JSON RPC Api](https://github.com/badaix/snapcast/blob/master/doc/json_rpc_api/v2_0_0.md). Regarding the command line: Pull requests are welcome.

daredoes, 2023-05-23:
I'll look into updating the client. Love snapcast, glad to be able to try and give back