iotsound/iotsound #350: DAC Bonnet Cyclical Static

BMW4673, 2020-11-18:
My Raspberry Pi3 with DAC Bonnet produces static pulsing 5 times about every 90 seconds. I’ve switched power strips and reimaged memory. I’ve checked all connections, resat bonnet and have not soldered anything onto the board. Shareport-Sync is running but this occurs when it’s idle. What else can I do?

BMW4673, 2020-11-23:
I rebuilt my Pi 3+ yesterday reseating the DAC and sequentially building the Pi. Again the issue demonstrated itself once the DAC Bonnet was installed. Is there anyone else observing the same error?

shawaj, 2020-12-26:
My first thought is - how are you sending music to the device? Is it over Bluetooth? Maybe it could be a similar issue to this https://github.com/raspberrypi/linux/issues/1402

Have you spoken to @adafruit about this issue also? Maybe they might be able to help more with it.

BMW4673, 2020-12-30:
@AlexProgrammerDE I am sending music via WiFi using Shairport. It has been identified as a bandwidth issue. Migrating from 2.4ghz to 5ghz, increasing transmit strength of my router and increasing error correction of the software contributing to significantly mitigating the static. I've similarly opened a case with Adafruit, but have never received a response.

shawaj, 2020-12-30:
@BMW4673 strange. It's weird that it happens when it's idle also.

It sounds maybe like the WiFi is causing interference on the audio somewhere.

I'm not really sure it could be a bandwidth issue because you said it only happens when the DAC bonnet is attached. 

I am not sure if any of our team have this board available for testing to see if we can reproduce the issue. I will ask.

In the meantime - could you post a picture of how you have the bonnet connected? Do you have a stacking header between the Raspberry Pi and the DAC?

BMW4673, 2021-01-01:
I am using a stacking header between the Raspberry PI and the DAC. I have a second Raspberry PI running the same image, however it does not have the DAC; and there is no static with that. I've swapped the media between the two, and the static remains on the Raspberry Pi w/ DAC. Unfortunately I'm not motivated enough at this moment to take about the radio where the Pi w/ DAC is installed to take photos.

shawaj, 2021-01-01:
@BMW4673 yeah that sounds to me as if the WiFi is interfering with the DAC bonnet potentially.

Are you able to try plugging it in with Ethernet instead? If you plug in the ethernet cable, and then from a terminal on the Host OS in balenaCloud disable the wifi using the command `ifconfig wlan0 down`

And then see if you still have the static issue?

You could also try adding a dtoverlay of "disable-bt" as well so it would be like this 

"hifiberry-dac","i2s-mmap","disable-bt" in the overlays section under "device configuration" menu item:

[image]

And you can add `SOUND_DISABLE_BLUETOOTH` under device variables menu item also:

[image]

BMW4673, 2021-06-15:
RESOLVED - this is not an issue with Shairport, but with the DAC Bonnet code. Per [I2S Speaker Bonnet Sound Quality](https://forums.adafruit.com/viewtopic.php?f=19&t=143946&start=30) discussion, when installing the scripts to enable the bonnet, selecting "no" to the recommended eliminating of static conversely eliminates the static and popping. There's only a single configuration question in the script, so that's what you're looking to answer no to. On two different PiZero WH with DAC Bonnet, as soon as I eliminated the dev/zero in the configuration the static ceased upon restart.

closed 2021-06-15 by BMW4673