iotsound/iotsound #353: Bluetooth not discoverable by some Devices (LG WebOS, FireTVStick)

XRTce, 2020-11-19:
When trying to connect a LG WebOS TV or an Amazon Fire TV Stick the Balena Sound device does not show up in the buetooth devices list. Connecting via iPhone works as expected.

Steps to reproduce the behavior:
1. Unpair any connected devices from the Balena Sound device
2. On TV Stick or LG TV go to the Settings and try to pair a new Bluetooth device.
3. The Balena Sound device does not show up in the device list

**Expected behavior**
The Balena Sound device obviously should show up in the devices list

**Device information**
 - Board (hardware info): Raspberry Pi 3
 - Device type (as seen in balenaCloud): Raspberry Pi 3
 - balenaOS version: balenaOS 2.60.1+rev1

fred771, 2021-01-11:
Hello all,
I have the same issue on balenaOS 2.65.0+rev1 version with LG TV

DBHey, 2021-02-05:
Hello,

Same issue as well, LG HF80LSR Projector with rPi 3B+ and balenaOS 2.67.3+rev4

youngd24, 2021-06-14:
Same issue on 2 FireTV sticks, both newer hardware and current updates. BT and Spotify connect work, just not the FireTV. Pi 3B+ and a Geekworm X400 (suptronics) DAC.

sukram911, 2021-08-09:
Hello!

Same problem here LG webOS TV UK6950PLB and Raspberry pi4 with balenaOS 2.82.10+rev1 . Phone Bluetooth and Spotify works well but tv won't find Balena. But TV finds Xiaomi Airdots

Webatron11, 2021-09-08:
Im also having this problem, Samsung TV and balenaOS 2.80.3+rev3 and RPi3

abhatikar, 2022-01-30:
Any update on this issue?

2mcreations, 2022-02-06:
i have same problem whit raspberry pi 3 x64 and fire stick. I have try whit bt usb external but, is the same

KoToZoid, 2022-02-10:
Same problem with TV Box Xiaomi MIBOX4.
Raspberry Pi 4 (using 64bit OS), balenaOS 2.88.5+rev1. 
Phone Bluetooth and Spotify works well but TV Box couldn't find Balena device. (This TV Box works well with Avantree Bluetooth Transmitter/Receiver.)

KoToZoid, 2022-02-11:
> Same problem with TV Box Xiaomi MIBOX4. Raspberry Pi 4 (using 64bit OS), balenaOS 2.88.5+rev1. 

I solved it for my setting this way. 
1. Install any Bluetooth Scanner on TV Box (I use "Bluetooth Scanner for Android TV" by ZDWorks Inc. https://play.google.com/store/apps/details?id=com.zdworksinc.bluetoothtvscanner&hl=ru&gl=US) 

2. Set Device variable BLUETOOTH_PAIRING_MODE for bluetooth service to LEGACY (as described in https://github.com/balenablocks/bluetooth) and restart bluetooth service. SSP mode was not helpful fro me.
If you like you can set BLUETOOTH_DEVICE_NAME, it's optional.

3. Search Raspberry from TV Box by Bluetooth Scanner and pair. (Default PIN code for legacy mode pairing is 0000. It also could be changed by variables, but I haven't done it.) You can see request for PIN code in logs of bluetooth service at balena dashboard.

4. There is still no connection after successful pairing. It's ok. Just restart  bluetooth service for connection. You can see in logs something like: 
\- Attempting to reconnect to MIBOX (**:**:**:**:**:**)...
\- Successfully connected to MIBOX  (**:**:**:**:**:**)...

Now it is connected and works well!

abhatikar, 2022-02-19:
Firetv does not have a app for Bluetooth scanning.

2mcreations, 2022-02-19:
I tried with the file viewer app, with firestick it has the transfer function bluetooth, but even in this case it can't connect

jonaskor, 2022-08-07:
Has anyone figuered out a solution for this yet? I would love to use the balenasound for streaming the audio of my FireTV stick!

jchisholm59, 2023-01-20:
Yes.
This works perfectly.
[
](https://www.amazon.ca/Bluetooth-Transmitter-Receiver-Wireless-Adapter/dp/B08K8JT6DR/ref=sr_1_5?crid=9MBLHZH2C3AX&keywords=taotronics+bluetooth&qid=1674251568&sprefix=taotronics+bluetooth%2Caps%2C79&sr=8-5)

jchisholm59, 2023-01-20:
Yes.
Just get a Bluetooth receiver/transmitter from Amazon such as the Taotronics gadget for 20 bucks.
Works perfectly.