iotsound/iotsound #355: Add Support for RaspiAudio Audio+ DAC HAT

paulhennings, 2020-11-22:
I am running a Pi Zero W with the "RaspiAudio Audio+" DAC.
No sound comes out of my hat (3.5 mm jack).

This is the website for the DAC:
https://raspiaudio.com/produit/audio
https://forum.raspiaudio.com/t/audio-installation-guide/19

RaspiAudio Audio+ DAC is not on the list of "DACs that are known to work with balenaSound" at:
https://sound.balenalabs.io/docs/audio-interfaces/#dac-boards

Please consider adding support for RaspiAudio Audio+ DAC. Thank you!

lud0182, 2020-12-12:
Hi,
I have the same problem :
+ Raspberry Pi 3B+
+ Audio DAC from raspiaudio Audio+
Thank you for considering our request.

shawaj, 2020-12-26:
@paulhennings @lud0182 I believe this DAC will work fine with balenaSound already, looking at the documentation.

You should be able to go to the device in question, click on "Device Configuration" and you will see a line that says "Define DT overlays" - you need to edit this and add "hifiberry-dac"

Also, just above this, in the DT params section, if it has an entry saying "audio=on" you need to change it to "audio=off"

paulhennings, 2020-12-27:
@shawaj @lud0182 Success! Thank you for this.
FYI, I had tried hifiberry-dac before. The missing key was setting audio=off.
Happy New Year!

shawaj, 2020-12-27:
@paulhennings fantastic that's great news! The audio=off part basically turns the onboard audio off so is essential otherwise the DAC and onboard audio conflict with each other.

I'll get it added to the list of known working DACs at some point then. For now please leave this issue open so we don't forget.

And happy new year to you too 🎉

closed 2021-01-01 by bulldozer-balena[bot]

MTolba, 2021-03-11:
It might be a good idea to mention the name of the DAC explicitly in the docs because this confused some users on the forums.

PR #389 (2021-01-01): docs: add various dacs that have been tested
adds instructions for various dacs that have been tested to work with balenaSound

Closes: #385 
Closes: #355 
Closes: #345 
Change-type: patch
commit 99d2e787 (2021-01-01): docs: add various dacs that have been tested

adds instructions for various dacs that have been tested to work with balenaSound

Closes: #385 
Closes: #355 
Closes: #345 
Change-type: patch