iotsound/iotsound #357: Pi 4B not working with Hifiberry DAC2 HD

richardpringle, 2020-11-24:
I had everything working beautifully before plugging in the DAC. 

- Raspberry Pi 4 Model B
- Hifiberry DAC2 HD
- balenaOS 2.60.1+rev5

I tried some things from #296 to no avail. The aux output isn't working either anymore.

Here are the logs from the `audio` container:
```
23.11.20 22:47:57 (-0500)  audio  
23.11.20 22:47:57 (-0500)  audio  --- Audio ---
23.11.20 22:47:57 (-0500)  audio  Starting audio service with settings:
23.11.20 22:47:57 (-0500)  audio  - Pulse log level: NOTICE
23.11.20 22:47:57 (-0500)  audio  - Default output: DAC
23.11.20 22:47:57 (-0500)  audio  
23.11.20 22:47:57 (-0500)  audio  Detected audio cards:
23.11.20 22:47:57 (-0500)  audio  0 Headphones bcm2835_headphonbcm2835Headphones-bcm2835Headphones
23.11.20 22:47:57 (-0500)  audio  1 sndrpihifiberry HifiberryDacplu-snd_rpi_hifiberry_dacplushd
23.11.20 22:47:57 (-0500)  audio  
23.11.20 22:47:57 (-0500)  audio  
23.11.20 22:48:26 (-0500)  audio  Setting audio routing rules. Note that this can be changed after startup.
23.11.20 22:48:26 (-0500)  audio  Routing 'balena-sound.input' to 'balena-sound.output'.
23.11.20 22:48:26 (-0500)  audio  Routing 'balena-sound.output' to 'alsa_output.dac.stereo-fallback'.
23.11.20 22:48:26 (-0500)  audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
23.11.20 22:48:27 (-0500)  audio  E: [pulseaudio] module-loopback.c: No such sink.
23.11.20 22:48:27 (-0500)  audio  E: [pulseaudio] module.c: Failed to load module "module-loopback" (argument: "source="balena-sound.output.monitor" sink="alsa_output.dac.stereo-fallback""): initialization failed.
23.11.20 22:48:27 (-0500)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
23.11.20 22:48:27 (-0500)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
23.11.20 22:48:27 (-0500)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
```

The audio jack doesn't seem to work either.

The output from running `pactl list sinks` in the `audio` container is as follows: 
```
root@bec19b8c0a64:/usr/src# pactl list sinks
Sink #0
        State: SUSPENDED
        Name: alsa_output.bcm2835.stereo-fallback
        Description: Built-in Audio Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 6
        Mute: no
        Volume: front-left: 56210 /  86% / -4.00 dB,   front-right: 56210 /  86% / -4.00 dB
                balance 0.00
        Base Volume: 56210 /  86% / -4.00 dB
        Monitor Source: alsa_output.bcm2835.stereo-fallback.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY 
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "bcm2835 Headphones"
                alsa.id = "bcm2835 Headphones"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "bcm2835 Headphones"
                alsa.long_card_name = "bcm2835 Headphones"
                alsa.driver_name = "snd_bcm2835"
                device.bus_path = "/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0"
                sysfs.path = "/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0"
                device.form_factor = "internal"
[... 1003 lines cut ...]
         "format" : 3,
         "rate" : 44100,
         "channels" : 2
      },
      "cookie" : 4095909592,
      "defaultSink" : "balena-sound.input"
   }
}
```

And running `dmesg | grep hifi` 
```bash
root@bec19b8c0a64:/usr/src# dmesg | grep hifi
[    8.980998] snd-rpi-hifiberry-dacplushd soc:sound: pcm179x-hifi <-> fe203000.i2s mapping ok
[    8.989655] snd-rpi-hifiberry-dacplushd soc:sound: ASoC: no DMI vendor name!
```

and finally `dmesg | grep snd` 
```bash
root@bec19b8c0a64:/usr/src# dmesg | grep snd
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:3A:4C:B2 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 rootfstype=ext4 rootwait root=UUID=ba1eadef-f68c-4f2e-a845-a0ee01c229a1 rootwait
[    8.808429] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    8.980998] snd-rpi-hifiberry-dacplushd soc:sound: pcm179x-hifi <-> fe203000.i2s mapping ok
[    8.989655] snd-rpi-hifiberry-dacplushd soc:sound: ASoC: no DMI vendor name!
```

richardpringle, 2020-11-24:
(I have two Pis and two DACs so I don't think any of the hardware is faulty)

richardpringle, 2020-11-24:
To update: when I hardcode `alsa_output.1.stereo-fallback` in the `start.sh` script, everything works perfectly (although I still get one of those `24.11.20 13:39:49 (-0500)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.` instead of two). 

I'm actually not quite sure what happens when I use the one-click-deploy with Balena-cloud. Before manually pushing edits, would Balena-Cloud have picked up new versions from this repo? Or would I have had to upgrade manually anyway?

richardpringle, 2020-11-24:
And another question (feel free to get me to write this in a forum somewhere)... are there more configuration options?
I see from this
```
Sink #1
        State: SUSPENDED
        Name: alsa_output.1.stereo-fallback
        Description: Built-in Audio Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 7
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.1.stereo-fallback.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY 
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "HiFiBerry DAC+ HD HiFi pcm179x-hifi-0"
                alsa.id = "HiFiBerry DAC+ HD HiFi pcm179x-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "1"
                alsa.card_name = "snd_rpi_hifiberry_dacplushd"
                alsa.long_card_name = "snd_rpi_hifiberry_dacplushd"
                alsa.driver_name = "snd_soc_hifiberry_dacplushd"
                device.bus_path = "/devices/platform/soc/soc:sound/sound/card1"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card1"
                device.form_factor = "internal"
                device.string = "hw:1"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.profile.name = "stereo-fallback"
                device.profile.description = "Stereo"
                device.description = "Built-in Audio Stereo"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Ports:
                analog-output: Analog Output (priority: 9900)
        Active Port: analog-output
        Formats:
                pcm
```
that the resolution is 16bits and the sample rate is configured to 44.1kHz. From the datasheet, it looks like this board can handle a resolution of 24bits and a sample rate all the way up to 192kHz, is it possible to change these values?

shawaj, 2020-12-26:
@richardpringle did you change the dtparams to say audio=off instead of audio=on ?

aandroide, 2020-12-27:
i bet it's the latest pulseaudio update causing the problem. you can try to downgrade pulseaudio and restart:
sudo apt install pulseaudio=12.2-4+deb10u1 libpulse0=12.2-4+deb10u1 pulseaudio-utils=12.2-4+deb10u1 libpulsedsp=12.2-4+deb10u1

richardpringle, 2020-12-29:
@shawaj, tried, didn't help 

@aandroide, I could downgrade, but I that's just as much effort as hardcoding the name of the Sink

shawaj, 2020-12-29:
@richardpringle with the audio=off set is it still sink 1 not sink 0?

shawaj, 2020-12-29:
Aha I've spotted the issue I think.

The udev rules don't currently have an entry for `snd-rpi-hifiberry-dacplushd`
https://github.com/balenablocks/audio/blob/master/udev/95-balena-audio.rules

Relates to #296 

Will submit a PR to fix this later.

shawaj, 2020-12-29:
PR is here https://github.com/balenablocks/audio/pull/48

shawaj, 2020-12-29:
@richardpringle until the PR is merged, you should be able to get this working by following the instructions here https://github.com/balenalabs/balena-sound/issues/377#issuecomment-752152387

shawaj, 2021-01-01:
Duplicate of #377

shawaj, 2021-01-01:
Closing this issue now in favour of #377 

Necessary changes were pulled to the audio block in https://github.com/balenablocks/audio/pull/49

You can update by using `git pull` and `balena push -c` to rebuild if using balena CLI to build again without using cached images.

closed 2021-01-01 by shawaj