iotsound/iotsound #367: Storage space filling up quickly

marevalod, 2020-12-03:
Hello,

First of all a big thank you to the developers. This project is really great, works perfectly and it's so easy to install! Really good work.

I've been using this for the past few weeks and I recently started to notice the audio stops after a while after playing with no issues (usually in the middle of the first song I play or at the beginning of the second one). It just stops playing and Spotify seems to "crash" or get confused about what's happening. I realized by looking at my dashboard that the disk space was full, which surprised me, as I'm pretty sure I had in excess of 2gb free when I installed the system (only a couple weeks ago - 8gb SD card). I'm wondering what's eating up space so quickly? 

One theory is Spotify (which is the only service I'm using really) is caching a lot of stuff and that is taking up the free space, but it doesn't seem realistic to have that amount of data as cache. Then I thought it might be the "download" feature Spotify has, where you can download playlists to your devices to listen to them when offline. I have this switched on for some of my playlists, and I believe it will download the tracks across all devices. I'm not sure if this would also happen on the Spotify service used for this deployment, but it would explain the rapid storage consumption.

Any advice or information would be greatly appreciated. I can always flash again and start from scratch, but I'm hoping to learn what's going on and also avoid reconfiguring everything again. It was running perfectly before this happened.

Thanks!

ssieb, 2020-12-03:
Why don't you investigate where the space is being used?  `du` is a very handy tool for that.  Depending on how you're running things, check in /var, /home, and /root to start with.  For a start, try `du -hxd1 /` and then work your way down.  e.g. `du -hxd1 /var`

newgas, 2020-12-03:
I have just recently encountered the exact same problem, though I have a 32GB card so it took many hours of streaming before I encountered the error, that is, first song stops. Tried to update to latest build and download kept failing, then I noticed there was no storage.

Looking into this further, i believe it is related to this https://github.com/dtcooper/raspotify/issues/144.

A quick fix that got me back to normal, was to "Purge data" in the console.

marevalod, 2020-12-03:
> Why don't you investigate where the space is being used? `du` is a very handy tool for that. Depending on how you're running things, check in /var, /home, and /root to start with. For a start, try `du -hxd1 /` and then work your way down. e.g. `du -hxd1 /var`

Thanks for the suggestion, I ran the commands but don't see anything too suspicious. Do you?

**du -hxd1 /**
`3.0K    /quirks
93M     /usr
2.0K    /mnt
1.0K    /media
1.3M    /etc
3.2M    /bin
19M     /boot
2.0K    /home
79M     /lib
8.2M    /resin-boot
3.0K    /var
1.8M    /sbin
205M    /`

**du -hxd1 /var**
`1.0K    /var/local
1.0K    /var/backups
3.0K    /var`

**du -hxd1 /usr**
`1.0K    /usr/src
1.0K    /usr/include
2.1M    /usr/libexec
57M     /usr/bin
1.0K    /usr/games
25M     /usr/lib
6.7M    /usr/sbin
2.7M    /usr/share
93M     /usr`

**du -hxd1 /var/cache**
`0       /var/cache/private
0       /var/cache/opkg
8.0K    /var/cache/ldconfig
8.0K    /var/cache`

> I have just recently encountered the exact same problem, though I have a 32GB card so it took many hours of streaming before I encountered the error, that is, first song stops. Tried to update to latest build and download kept failing, then I noticed there was no storage.
> 
> Looking into this further, i believe it is related to this [dtcooper/raspotify#144](https://github.com/dtcooper/raspotify/issues/144).
> 
> A quick fix that got me back to normal, was to "Purge data" in the console.

Thanks for this suggestion too, I think we are facing the same issue. I tried purging data but it only freed up a couple hundred MB. I think this is more like a couple GB kind of problem.

Thanks all!

eiddor, 2020-12-03:
@marevalod There are a few different mounts in balenaSound.  Can you post a `df -h` from the Host container so we can see exactly which fs is being filled?

Here's what mine looks like:

```
root@c2762a7:~# df -h
Filesystem                      Size  Used Avail Use% Mounted on
devtmpfs                        1.8G     0  1.8G   0% /dev
tmpfs                           1.9G  4.0K  1.9G   1% /tmp
/dev/mmcblk0p3                  300M  292M     0 100% /mnt/sysroot/active
/dev/disk/by-state/resin-state   19M  244K   17M   2% /mnt/state
overlay                         300M  292M     0 100% /
/dev/mmcblk0p6                   58G  1.6G   54G   3% /mnt/data
tmpfs                           1.9G     0  1.9G   0% /dev/shm
tmpfs                           1.9G   22M  1.9G   2% /run
tmpfs                           1.9G     0  1.9G   0% /sys/fs/cgroup
tmpfs                           1.9G   20K  1.9G   1% /var/volatile
/dev/mmcblk0p1                   40M  7.4M   33M  19% /mnt/boot
/dev/mmcblk0p2                  300M  283M     0 100% /mnt/sysroot/inactive
```

Then we can cd to a directory that seems to be taking a lot of space and issue a:

`du | sort -n -r | head -n 10`

(This is one of my favorite commands and I have it aliased to "hog" on all of my Linux machines.)

I suspect the majority of your usage is going to be in `/mnt/data` which is where all of the container volumes live.

marevalod, 2020-12-03:
> @marevalod There are a few different mounts in balenaSound. Can you post a `df -h` from the Host container so we can see exactly which fs is being filled?
>
> Here's what mine looks like:
>
> ``` [...]

Thanks @eiddor. See below:

****
```
Filesystem                      Size  Used Avail Use% Mounted on
devtmpfs                        480M     0  480M   0% /dev
tmpfs                           487M     0  487M   0% /tmp
/dev/mmcblk0p2                  300M  225M   55M  81% /mnt/sysroot/active
/dev/disk/by-state/resin-state   19M  242K   17M   2% /mnt/state
none                            300M  225M   55M  81% /
/dev/mmcblk0p6                  6.6G  6.6G     0 100% /mnt/data
tmpfs                           487M  1.4M  486M   1% /dev/shm
tmpfs                           487M   38M  450M   8% /run
tmpfs                           487M     0  487M   0% /sys/fs/cgroup
tmpfs                           487M   20K  487M   1% /var/volatile
/dev/mmcblk0p1                   40M  8.3M   32M  21% /mnt/boot
/dev/mmcblk0p3                  300M  2.1M  278M   1% /mnt/sysroot/inactive
```

Your suspicion was right. `/mnt/data` has 6.6G/6.6G used. I ran the "hog" command and this is the output:
```
6842308 .
6842204 ./docker
5398580 ./docker/volumes
5398544 ./docker/volumes/1765536_spotifycache
5398540 ./docker/volumes/1765536_spotifycache/_data
5398528 ./docker/volumes/1765536_spotifycache/_data/files
1435000 ./docker/aufs
1433864 ./docker/aufs/diff
150024  ./docker/aufs/diff/941a6cd4ed774f6fbb599a10e20b6af554477ae768ae7b18cc49c47f57399b76
127364  ./docker/aufs/diff/7f6ba4542010d7b7711edd2f3e439e1a28829ea009c0c0abd0866454842e41a5
```

Please forgive my ignorance, but what did that command do? I still see the disk full. Should this have fixed the issue?

What would a permanent solution be to avoid this from keep happening over time?

Thank you all!

eiddor, 2020-12-03:
@marevalod Ok, so as you suspected, it's Spotify's cache that is taking up all of your space.

(The hog alias just take the du output and sorts it to show you the top ten directories in terms of space usage.  It's nothing fancy, just an easier way to figure out what's filling your file systems.)

So, just a guess, but I'm thinking that you're also right about the Spotify download function - Do you have your Spotify credentials configured?  I believe they're only required if you want to use Spotify Connect over the Internet.  If you're using Spotify Connect locally then you don't need them.  I don't use Spotify a whole lot with my balenaSound, so I can't really compare.

If it's not the download function/credentials, then it might just be normal operation and you need a larger microSD.

eiddor, 2020-12-03:
Not there are a couple of suggestions in the raspotify issue linked above that can also help you.  

Looking at `/etc/default/raspotify` in the Spotify container I see my cache as disabled:

```
root@c2762a7:/etc/default# cat raspotify | grep CACHE
#CACHE_ARGS="--cache /var/cache/raspotify"
```

And my cache directory usage is nearly nil:

```
root@c2762a7:~# du -h /var/cache/raspotify
1.0K    /var/cache/raspotify/files
2.0K    /var/cache/raspotify
```
 
I don't want to send you too far down the raspotify rathole, but these are things you should be able to check pretty quickly.

marevalod, 2020-12-03:
> @marevalod Ok, so as you suspected, it's Spotify's cache that is taking up all of your space.
>
> (The hog alias just take the du output and sorts it to show you the top ten directories in terms of [...]

That's a good point. I don't have my credentials configured, so probably the download feature wouldn't be in effect, right? 
It still sounds crazy to me that several GB of data was cached only over a couple of weeks. This doesn't sound like normal behavior, but I might be wrong.

Anyways, how do I solve this!? I guess there has to be a variable to limit cache size or a way to establish a cache wipe periodically? I don't know... I can't be the only one experiencing this issue!

Thanks again!

eiddor, 2020-12-04:
@marevalod See my last post for a couple of times - There are also a few different ideas in this thread: https://github.com/dtcooper/raspotify/issues/144

If I get a chance this weekend I'll try to play some Spotify on mine to see if the usage increases.

marevalod, 2020-12-04:
> @marevalod See my last post for a couple of times - There are also a few different ideas in this thread: [dtcooper/raspotify#144](https://github.com/dtcooper/raspotify/issues/144)
> 
> If I get a chance this weekend I'll try to play some Spotify on mine to see if the usage increases.

Thanks @eiddor. I can't seem to replicate what you did there. Those directories don't exist on my system apparently. What am I doing wrong?

eiddor, 2020-12-04:
> Thanks @eiddor. I can't seem to replicate what you did there. Those directories don't exist on my system apparently. What am I doing wrong?

Are you in the Spotify container?

FWIW - I played a few songs on mine last night and I can see my cache directory filling up, so this appears to be normal.

The thread I linked above has some tips for space maintenance - There may be some cache limit but it might be larger than your microSD and I'm not sure if it's worth trying to change the setting with a workaround.  

To be honest, I think the "fix" in your case is to use a 16G card, but that's up to you.

marevalod, 2020-12-04:
> Are you in the Spotify container?
>
> FWIW - I played a few songs on mine last night and I can see my cache directory filling up, so this appears to be normal.
>
> The thread I linked above has some tips [...]

I wasn't on the Spotify container, thanks for that. Now I can see the output and yes, it's the same as yours. 

Thanks for trying this thing out yourself @eiddor, it's good to know it's "normal" and not some problem with my setting. I can switch the micro SD no problem, but I think it'd be nice to see this as a configurable parameter, maybe toggle cache on/off, or determine cache size limit. Also, I didn't find mentions referring to the required micro SD size, and at least in my case, an 8GB micro SD is simply not working after it gets filled up with normal use.

eiddor, 2020-12-04:
@marevalod You're not wrong - In fact, the docs say that 8GB should be plenty enough:

[image]

So I think we have a couple of options - I'm not a dev or even dev-adjacent, so these are just ideas:

Option #1 

- Upgrade to 16GB and run for a while to see what happens.  There may be a built-in cache limit that will fit on on a 16GB card.
- If so, update the docs to recommend 16GB
- If not, implement one of the workarounds in the threads above (there seem to be a few of them)

Option #2 

- Figure out the how to set the librespot option discussed in [this thread](https://github.com/plietar/librespot/pull/204) and add it as an environment variable in balenaSound

newgas, 2020-12-04:
The size of the card just delays the problem, but will help.

As I posted earlier, I experienced the exact same issue with a 32gb card. Purged data less than a week ago and it is now slowly filling up again. I only use Spotify connect.

It will depend on your listening habits, as I assume that listening to the same tracks or albums, they will be retrieved from the cache compared to random songs and playlists where the cache will keep building.

eiddor, 2020-12-05:
So, I did some more digging - If you edit ```plugins/spotify/start.sh``` in the balenaSound package and add ```--disable-audio-cache``` to the librespot command, this might disable the cache.

eg.

```
set -- /usr/bin/librespot \
  --name "$SOUND_DEVICE_NAME" \
  --bitrate "$SOUND_SPOTIFY_BITRATE" \
  --cache /var/cache/raspotify \
  --disable-audio-cache \
  --linear-volume \
  "$@"
```

Someone smarter than me can probably modify that script to allow for an environment variable in balena to turn this on or off.

marevalod, 2020-12-07:
> So, I did some more digging - If you edit `plugins/spotify/start.sh` in the balenaSound package and add `--disable-audio-cache` to the librespot command, this might disable the cache.
>
> eg.
>
> ```
> set [...]

Good find @eiddor. I'll try that as well. Thanks!

eiddor, 2020-12-08:
@marevalod I just tested it out here and it works as expected.  You may still want to clear out the ```/var/cache/raspotify/files``` directory in the spotify container just to get rid of the old files, or kill the ```spotifycache``` volume at the Docker level if you're comfortable doing so.

eiddor, 2020-12-10:
@marevalod I decided to do a bit of messing around/learning and came up with a fix and submitted a PR to add an option to disable the Spotify audio cache.  

Assuming it's approved and merged, you'll be able to enable ```SOUND_SPOTIFY_DISABLE_CACHE``` as an environment variable (set to anything), which will disable the cache.

There are a few ways to delete the existing cache, but the most thorough way would be to delete the Docker volume itself before pushing the new version.

In a session of the Host OS:

1) Get the Container ID of the Spotify container with ```balena ps | grep spotify``` (it's the first field, ```f0b85737f9a9``` in my case)

```
root@c2762a7:~# balena ps | grep spoti
**f0b85737f9a9**        e619eb7be755                         "/usr/bin/entry.sh /…"   17 hours ago        Up 17 hours                                      spotify_3043564_1625757
```

2) Stop the Spotify container with ```balena stop <container id>```

3) Remove the Spotify container with ```balena rm <container id>```

4) Get the Volume name with ```balena volume ls``` (it's ```1751225_spotifycache``` in my case).

```
root@c2762a7:~# balena volume ls
DRIVER              VOLUME NAME
local               **1751225_spotifycache**
```

5) Remove the volume with ```balena volume rm <volume name>```

6) Push new version of balenaSound

I hope that helps!
Roddie

marevalod, 2020-12-23:
> @marevalod I decided to do a bit of messing around/learning and came up with a fix and submitted a PR to add an option to disable the Spotify audio cache.
> 
> Assuming it's approved and merged, you'll be able to enable `SOUND_SPOTIFY_DISABLE_CACHE` as an environment variable (set to anything), which will disable the cache.
> 
> There are a few ways to delete the existing cache, but the most thorough way would be to delete the Docker volume itself before pushing the new version.
> 
> In a session of the Host OS:
> 
> 1. Get the Container ID of the Spotify container with `balena ps | grep spotify` (it's the first field, `f0b85737f9a9` in my case)
> 
> ```
> root@c2762a7:~# balena ps | grep spoti
> f0b85737f9a9        e619eb7be755                         "/usr/bin/entry.sh /…"   17 hours ago        Up 17 hours                                      spotify_3043564_1625757
> ```
> 
> 1. Stop the Spotify container with `balena stop <container id>`
> 2. Remove the Spotify container with `balena rm <container id>`
> 3. Get the Volume name with `balena volume ls` (it's `1751225_spotifycache` in my case).
> 
> ```
> root@c2762a7:~# balena volume ls
> DRIVER              VOLUME NAME
> local               1751225_spotifycache
> ```
> 
> 1. Remove the volume with `balena volume rm <volume name>`
> 2. Push new version of balenaSound
> 
> I hope that helps!
> Roddie

Thanks @eiddor. Great job! Let's hope this gets approved so we can use the feature from now on.

tmigone, 2021-01-19:
Hey @eiddor @marevalod, thanks so much for the investigation + PR, you guys were spot on with it. Most of my SD cards are either 32GB or 64GB so it makes sense I never encountered this problem while developing/testing balenaSound, so huge thanks to both of you.

@marevalod is right, we clearly state an 8GB is enough and I'd like to stick to that for as long as possible. 

I'll review the PR now so we can get it merged asap.

eiddor, 2021-01-19:
Thanks, @tmigone - Welcome back!

tmigone, 2021-05-27:
Doing some cleanup, this was addressed by #369 a while back. Closing!

closed 2021-05-27 by tmigone