iotsound/iotsound #368: Docs do not specify RAM requirements

gewrites, 2020-12-04:
I'm not sure how much RAM is needed in a Raspberry Pi to run this software.

shawaj, 2020-12-25:
balenaSound will work very nicely with a Pi Zero W - which has 512MB of RAM.

So basically, any Raspberry Pi currently on the market will be fine.

Having said that - it depends what you want to do with the device. If it will just be used for balenaSound then the Pi Zero W would be great... But you will need to use a soundcard for it. If you want to use the device for balenaSound and other applications simultaneously you might want to purchase a more expensive one.

Hope that helps.

shawaj, 2020-12-26:
By the way @gewrites I forgot to say... With the pi zero, pi b, pi b+, pi 2b the multiroom client and multiroom server functionality are turned off by default so they will just work as standalone devices, unless you override this.

If you have a pi 3b, 3b+, 4b these work better for the multiroom server. The less powerful Pis may work ok as a client but may struggle as the server.

AlexProgrammerDE, 2020-12-29:
As @shawaj said. You can basicly use any raspberry pi with this.

closed 2020-12-29 by AlexProgrammerDE