iotsound/iotsound #371: Support to a Tidal Connect

nmcmartins, 2020-12-12:
An excellent improvement will be to include support for Tidal Connect.

shawaj, 2021-01-02:
More info on [Tidal connect](https://tidal.com/connect)

It appears to be closed source. You have to register your interest [here](https://airtable.com/shrjVuIJMWIG96uR9) and wait to hear back from them.

shawaj, 2021-01-02:
Just found this after some googling https://github.com/seniorgod/ifi-tidal

shawaj, 2021-01-03:
Have almost got this running as a plugin. Have pushed my dev branch here #399 

<details>
<summary>Full error output...</summary>

```
root@2ec9b5a:/usr/ifi/ifi-tidal-release/bin# /usr/ifi/ifi-tidal-release/bin/tidal_connect_application \
>                         --tc-certificate-path "/usr/ifi/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat" \
>                         --netif-for-deviceid wlan0 \
>                         -f "balenaSound stream to project" \
>                         --codec-mpegh true \
>                         --codec-mqa false \
>                         --model-name "balenaSound Streamer"
STARTING TidalConnect services, version: 1.1.3
[2021-01-03 21:19:00.254] [tisoc] [info] [HttpClient.cpp:25] certification path: /etc/ssl/certs
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib conf.c:5014:(snd_config_expand) Unknown parameters {AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[2021-01-03 21:19:00.409] [tisoc] [debug] [DeviceInfo.cpp:103] mac addr : "dc:a6:32:a4:af:4d"
[2021-01-03 21:19:00.409] [tisoc] [debug] [DeviceInfo.cpp:113] device id: "c263d56a5eb7cf4c137755762be5fb00"
[2021-01-03 21:19:00.409] [tisoc] [trace] [avahiImpl.cpp:33] serviceName : balenaSound Streamer-e322c9768461956dbefe40c0d0bae127
[2021-01-03 21:19:00.409] [tisoc] [info] [ServiceManager.cpp:19] service registered: clientcomm
[2021-01-03 21:19:00.409] [tisoc] [info] [ServiceManager.cpp:19] service registered: queue
[2021-01-03 21:19:00.409] [tisoc] [info] [ServiceManager.cpp:19] service registered: playback
[2021-01-03 21:19:00.409] [tisoc] [info] [ServiceManager.cpp:19] service registered: session
[... 41 lines cut ...]
[2021-01-03 21:19:00.473] [tisoc] [info] [ServiceManager.cpp:27] service started: session
[2021-01-03 21:19:00.473] [tisoc] [info] [ServiceManager.cpp:28] 

[... cut ...]
[2021-01-03 21:19:00.413] [tisoc] [error] [avahiImpl.cpp:70] avahi start failed.
```

Basically I think it is trying to use avahi / avahi-daemon - but it can't because balenaOS doesn't have systemd in the containers. So I think I maybe need to find some way to use DBUS to work with avahi in the balena HostOS.

Bit in over my head now to be honest! Will keep messing around with it.

But the good news is that the tidal connect service is authenticating and it is running - it just doesn't actually advertise properly over mDNS so the Tidal app can't find it.

shawaj, 2021-01-04:
@nmcmartins good news - i got it working https://github.com/balenalabs/balena-sound/pull/399#issuecomment-753708240

Need to tidy it up and sort out volume control and some other stuff but it is working so far.

npmstart-pray, 2021-09-06:
Any updates/progress reports?

marcgarciamarti, 2021-10-29:
> @nmcmartins good news - i got it working https://github.com/balenalabs/balena-sound/pull/399#issuecomment-753708240
> 
> 
> 
> Need to tidy it up and sort out volume control and some other stuff but it is working so far. If you want you can test it using balena CLI...

Hi! Could you provide guidance on how to enable this?

Thanks!!