iotsound/iotsound #375: Contraints for DAC boards

colepfwoerner, 2020-12-19:
Hello,

I love to see on your documentation that Balena sound supports these external DAC boards. I'm currently in the process of designing my own DAC board. I'm wondering, what are the constraints DAC boards should adhere to? for example: do they need to be stereo/mono? I've seen DAC chips with left and right justified I2S data, is that important? do I need to give chips a certain I2C address? Are there other considerations I might be missing?

My goal is to design a simple DAC board which could use one of the configuration settings for the already supported commercial DAC boards.

Thanks for the help,

CPFW

shawaj, 2020-12-25:
@colepfwoerner as a rule of thumb, if they work with Raspberry Pi they will be quite likely to work with balenaSound. The ones added to balenaSound, I believe, are just ones that we had the ability to test - i.e. one of our team actually have the board.

As you say though, if you look at one of the ones that's already supported and use the same or a compatible DAC chip that will be the easiest method for you.

For example the PCM5121, PCM5102 or PCM5122 from burr brown / Texas instruments is used in a lot of the boards - many of whom just piggy back off the hifiberry, JustBoom or iqaudio drivers...

shawaj, 2021-01-01:
@colepfwoerner do you need any further assistance here with anything?

colepfwoerner, 2021-01-03:
I think I've got it figured out. I went with the UDA 1334 chip which is one one of the Adafruit boards. They had a schematic on their website that showed how they had configured the connections, which is what I followed. I'll let you know if I have any issues once the board is assembled and plugged in.

A lot of the PCM-#### chips seem to require an external clock, which is not something I wanted to deal with.

colepfwoerner, 2021-01-03:
Thank you for the quick help!

closed 2021-01-03 by colepfwoerner

shawaj, 2021-01-04:
@colepfwoerner cool. If it is helpful here is a PCM5122 schematic...

[DAC_HAT_Schematic.PDF](https://github.com/balenalabs/balena-sound/files/5762505/DAC_HAT_Schematic.PDF)

And the corresponding pinout here https://pinout.xyz/pinout/dac_hat

You can see the bitclock on pin12. Basically, the PCM5122 and other compatible chips don't **need** an external clock - you can use one, but you can also just use the clock on the raspberry pi and use the pll clock divider stuff. Some of the more expensive raspberry pi dacs use an external clock - but the audio quality gains are negligible, and the average listener probably wouldn't notice the difference (in my opinion).

This is an interesting article about it https://hifiduino.wordpress.com/2014/11/13/raspberry-pi-b-digital-audio/

shawaj, 2021-01-04:
Also - feel free to let us know if you need any other help with getting your new dac working with balenaSound

colepfwoerner, 2021-01-04:
Very cool, Thank you