iotsound/iotsound #377: Hifiberry DAC HD2 no sound

arjan5340, 2020-12-23:
Dear,
I use a raspberry pi 4 and deploy with Balena. When i use the jack plug i have sound. When i install the hardware Hifiberry DAC HD2 on top of the raspberry and i make a new custom variable BALENA_HOST_CONFIG_dtoverlay with value hifiberry-dacplushd. Reboot the raspberry and i have no sound. Please can anybody told me what i doing wrong. Thanks

shawaj, 2020-12-25:
Just for reference this is a duplicate of #357

arjan5340, 2020-12-25:
Dear shawaj,

Yes i see i have explore the fully internet for looking how i can solve my problem so that also my Hifiberry dad hd2 card work. Please can you told me how i can solve my problem?
Please let me know

shawaj, 2020-12-25:
@arjan5340 could you report the output of the following commands from the terminal:

```
aplay -l
cat /proc/asound/modules
cat /proc/asound/cards
```

arjan5340, 2020-12-26:
Dear,

Thanks for your help. Hereby the information you ask for. Hopefully you will and can help me.

root@569d0a627462:/usr/src# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: bcm2835-jack [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
root@569d0a627462:/usr/src# 

root@569d0a627462:/usr/src# cat /proc/asound/modules
 0 snd_bcm2835

root@569d0a627462:/usr/src# cat /proc/asound/cards
 0 [bcm2835-jack   ]: bcm2835_headphonbcm2835 Headphones - bcm2835 Headphones
                      bcm2835 Headphones

shawaj, 2020-12-26:
@arjan5340 and this was from the `sound` container on balenaCloud or a different one?

It hasn't loaded the soundcard which is why you aren't getting any sound.

Let me know answer to my question above, and then I'll provide some screenshots for you of the settings you will need in balenaCloud to fix this.

arjan5340, 2020-12-26:
This screenshot is from my balena sound. 

Verstuurd vanaf mijn iPhone

shawaj, 2020-12-26:
@arjan5340 there was no picture? Can you try posting it directly on GitHub via the "view it on GitHub" link

arjan5340, 2020-12-26:
Sorry hereby the screenshot again. Thanks for your help iam new with Balena. I have 3 pi's and 3 amplifiers with on each amplifier 2 speakers. When i use the jack plug i have sound but when i plug on the Hifiberry DAC HD2 boards i don't have any sound. Its really fantastic when you will and can help me. 
[image]

shawaj, 2020-12-26:
@arjan5340 where it says "Define DT parameters" it currently says `"i2c_arm=on","spi=on","audio=on"` you need to change this to `"i2c_arm=on","spi=on","audio=off"`

Also you can delete the BALENA_HOST_CONFIG_dtoverlay declaration and above where it says "Define DT overlays" you can change this to `"vc4-fkms-v3d","hifiberry-dacplus"`

arjan5340, 2020-12-26:
Dear Shawaj,

I try it now. Many thanks for help.

shawaj, 2020-12-26:
@arjan5340 you are using the HD2 DAC right? If so the correct overlay for you is `hifiberry-dacplushd`

I updated the above instructions to reflect that

arjan5340, 2020-12-26:
you are using the HD2 DAC right? that is correct. i try hifiberry-dacplushd

arjan5340, 2020-12-26:
I have changed the parameters as you say. But i don't have any sound yet.

root@569d0a627462:/usr/src# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplushd], device 0: HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 [HiFiBerry DAC+ HD HiFi pcm179x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

root@569d0a627462:/usr/src# cat /proc/asound/modules
 0 snd_soc_hifiberry_dacplushd

root@569d0a627462:/usr/src# cat /proc/asound/cards
 0 [sndrpihifiberry]: HifiberryDacplu - snd_rpi_hifiberry_dacplushd
                      snd_rpi_hifiberry_dacplushd

In the screenshot the parameters as i set now:
[image]

arjan5340, 2020-12-26:
Dear shawaj

Hereby the desmet hope this help for you.

root@569d0a627462:/usr/src# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.58 (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 SMP PREEMPT Wed Dec 16 13:40:47 UTC 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1025536
[    0.000000]   DMA zone: 4096 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 255488 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 32 pages/cpu s90392 r8192 d32488 u131072
[    0.000000] pcpu-alloc: s90392 r8192 d32488 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1009408
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1  smsc95xx.macaddr=DC:A6:32:86:C4:5B vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 rootwait     cgroup_enable=memory root=UUID=ba1eadef-3804-4ba7-acaa-22c728d7b67d rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x3a600000-0x3e600000] (64MB)
[    0.000000] Memory: 3665924K/4102144K available (10748K kernel code, 1176K rwdata, 3812K rodata, 10560K init, 1223K bss, 174076K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36209 entries in 142 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x320/0x4ac with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000217] Console: colour dummy device 80x25
[    0.000622] printk: console [tty1] enabled
[    0.000680] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000713] pid_max: default: 32768 minimum: 301
[    0.001034] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001093] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.024025] ASID allocator initialised with 32768 entries
[    0.032017] rcu: Hierarchical SRCU implementation.
[    0.040379] EFI services will not be available.
[    0.048077] smp: Bringing up secondary CPUs ...
[    0.080272] Detected PIPT I-cache on CPU1
[    0.080329] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.112376] Detected PIPT I-cache on CPU2
[    0.112416] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.144484] Detected PIPT I-cache on CPU3
[    0.144523] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.144671] smp: Brought up 1 node, 4 CPUs
[    0.144756] SMP: Total of 4 processors activated.
[    0.144777] CPU features: detected: 32-bit EL0 Support
[    0.144798] CPU features: detected: CRC32 instructions
[    0.169025] CPU: All CPU(s) started at EL2
[    0.169091] alternatives: patching kernel code
[    0.170569] devtmpfs: initialized
[    0.180030] Enabled cp15_barrier support
[    0.180078] Enabled setend support
[    0.180672] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[... 292 lines cut ...]
[   36.117155] br-d511e847fed1: port 2(vethfa74183) entered forwarding state
[   36.480310] br-d511e847fed1: port 2(vethfa74183) entered disabled state
[   37.553295] eth0: renamed from vethe98f463
[   37.581499] IPv6: ADDRCONF(NETDEV_CHANGE): vethfa74183: link becomes ready
[   37.588719] br-d511e847fed1: port 2(vethfa74183) entered blocking state
[   37.595457] br-d511e847fed1: port 2(vethfa74183) entered forwarding state
[   38.665465] br-d511e847fed1: port 2(vethfa74183) entered disabled state
[   38.672481] vethe98f463: renamed from eth0
[   38.828315] br-d511e847fed1: port 2(vethfa74183) entered disabled state
[   38.851077] device vethfa74183 left promiscuous mode
[... cut ...]
[   39.680319] br-d511e847fed1: port 2(veth229fde3) entered disabled state
[   39.736310] Bluetooth: RFCOMM TTY layer initialized
[   39.741649] Bluetooth: RFCOMM socket layer initialized
[   39.747026] Bluetooth: RFCOMM ver 1.11
[   40.806218] eth0: renamed from veth2ab1aa7
[   40.829086] IPv6: ADDRCONF(NETDEV_CHANGE): veth229fde3: link becomes ready
[   40.836364] br-d511e847fed1: port 2(veth229fde3) entered blocking state
[   40.843090] br-d511e847fed1: port 2(veth229fde3) entered forwarding state
root@569d0a627462:/usr/src#

shawaj, 2020-12-26:
@arjan5340 well the good thing is it's actually recognising the soundcard now so that's progress at least!

How are you trying to play music? Through Bluetooth or Spotify or something else?

Can you open a terminal in the audio container and try `speaker-test -c2 -t wav`

Also can you try running `alsamixer` and see what the master gain is set to? Then press F6 and select your soundcard and see what the gains are set to there also.

If you could post some screenshots of the output as well that would be great!

shawaj, 2020-12-26:
@arjan5340 could you also try just rebooting the device from the balenaCloud device management page first?

arjan5340, 2020-12-27:
Dear Shawaj,

I use spotify as music streamer. When i try to do a speaker test no sound. 
i open the alfamixer and the DAC is not the default sound card. I have select the DAC as sound card but i can't save it. I sent you some screenshots hope you can resolve this. Between some changes i have done the speaker test but without any result. Als i have done the reboot from the management page without any result.

arjan5340, 2020-12-27:
Dear Shawaj,
I see this after the reboot:

c: Sink alsa_output.dac.stereo-fallback does not exist.
27.12.20 10:07:10 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
27.12.20 10:07:10 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
27.12.20 10:07:10 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.

shawaj, 2020-12-27:
@arjan5340 when you say it's not the default soundcard in alsamixer - what do you mean by this? What options are you presented with?

arjan5340, 2020-12-27:
Dear Shawaj,

I mean this as follow. When i open Alsamixer the card is Pulseaudio.
[image]

I can make the choice under F6 for my sound card and after select my sound card you see them.
[image]

You see my sound card is select.
[image]

But when i close Aslamixer and reopen Alsamixer my card is again Pulseaudio. Maybe it doesn't matter. 

Is this clear for you?

shawaj, 2020-12-27:
@arjan5340 from the second screenshot it appears that the standard audio card is still loading. Bcm2835 one.

It shouldn't be doing that with the audio=off setting.

You know the changes you made to this, was that on the device itself? Or on the whole application?

shawaj, 2020-12-27:
By the way, pulseaudio is correct one to be default. It is the system sound. So that's the master volume. It will always be that as default when you start to alsamixer

arjan5340, 2020-12-27:
I have make the changes in the device. (See screenshot) 
[image]

I have done a reboot again and again i see the message below.
c: Sink alsa_output.dac.stereo-fallback does not exist.
27.12.20 10:07:10 (+0100) audio E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
27.12.20 10:07:10 (+0100) audio E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
27.12.20 10:07:10 (+0100) audio E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.

Is this the problem?

shawaj, 2020-12-27:
@arjan5340 sorry it's a bit difficult to know because I don't have the exact soundcard here. So I'm guessing a bit. Also I can't find the exact pinout of the board either so there is maybe some setting I'm unaware of.

Do you specifically need UART for anything? If not, can you click the delete button next to the "Enable / Disable UART"?

Can you also go into device variables in the menu on the left hand side, and change the AUDIO_OUTPUT one to say DAC

shawaj, 2020-12-27:
I think this definitely relates to #296 and #357 looking at it in more detail.

Could you start a terminal in the audio container and run...
```
pactl list sinks
dmesg | grep hifi
dmesg | grep snd
aplay -lL
```

And paste the output of each of those here?

shawaj, 2020-12-27:
This might be interesting to try also https://github.com/balenalabs/balena-sound/issues/357#issuecomment-751471439

However, as I said I don't have a board to test it.

arjan5340, 2020-12-27:
I have made a fresh install of BalenaSound and start tomorrow from scratch. The first I do tomorrow is update the raspberry to the last kernel. Hope that make sense. I give you tomorrow a update.

arjan5340, 2020-12-28:
Dear shawaj,

I have start from scratch but that make not sense. Below the information you asking for.

root@b998d39a2892:/usr/src# pactl list sinks
Sink #0
        State: SUSPENDED
        Name: alsa_output.bcm2835-jack.stereo-fallback
        Description: Built-in Audio Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 6
        Mute: no
        Volume: front-left: 56210 /  86% / -4.00 dB,   front-right: 56210 /  86% / -4.00 dB
                balance 0.00
        Base Volume: 56210 /  86% / -4.00 dB
        Monitor Source: alsa_output.bcm2835-jack.stereo-fallback.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY 
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "bcm2835 Headphones"
                alsa.id = "bcm2835 Headphones"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "bcm2835 Headphones"
                alsa.long_card_name = "bcm2835 Headphones"
                alsa.driver_name = "snd_bcm2835"
                device.bus_path = "/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0"
                sysfs.path = "/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0"
                device.form_factor = "internal"
                device.string = "hw:0"
                device.buffering.buffer_size = "131072"
                device.buffering.fragment_size = "131072"
                device.access_mode = "mmap+timer"
                device.profile.name = "stereo-fallback"
                device.profile.description = "Stereo"
                device.description = "Built-in Audio Stereo"
                alsa.mixer_name = "Broadcom Mixer"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Ports:
                analog-output-headphones: Headphones (priority: 9900)
        Active Port: analog-output-headphones
        Formats:
                pcm

Sink #1
        State: SUSPENDED
        Name: alsa_output.1.stereo-fallback
        Description: Built-in Audio Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 7
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.1.stereo-fallback.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY 
        Properties:
[... 159 lines cut ...]
hw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplushd, HiFiBerry DAC+ HD HiFi pcm179x-hifi-0
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplushd, HiFiBerry DAC+ HD HiFi pcm179x-hifi-0
    Hardware device with all software conversions
usbstream:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplushd
    USB Stream Output
**** List of PLAYBACK Hardware Devices ****
card 0: bcm2835-jack [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: sndrpihifiberry [snd_rpi_hifiberry_dacplushd], device 0: HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 [HiFiBerry DAC+ HD HiFi pcm179x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Hope you can help me with this information

arjan5340, 2020-12-28:
Maybe this can help?
https://github.com/balenalabs/balena-sound/issues/312 
but how can i get the this new deployment?

shawaj, 2020-12-28:
@arjan5340 the weird thing is how it is still loading the bcm2835 system audio.

Are you sure you have the "audio=off" set in the dtparams section?

arjan5340, 2020-12-28:
Sorry iam change so much things the audio was in the above information still "on" i set it now to "off" i send you the new information

root@b998d39a2892:/usr/src# aplay -lL
default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplushd, HiFiBerry DAC+ HD HiFi pcm179x-hifi-0
    Default Audio Device
dmix:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplushd, HiFiBerry DAC+ HD HiFi pcm179x-hifi-0
    Direct sample mixing device
dsnoop:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplushd, HiFiBerry DAC+ HD HiFi pcm179x-hifi-0
    Direct sample snooping device
hw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplushd, HiFiBerry DAC+ HD HiFi pcm179x-hifi-0
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplushd, HiFiBerry DAC+ HD HiFi pcm179x-hifi-0
    Hardware device with all software conversions
usbstream:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplushd
    USB Stream Output
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplushd], device 0: HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 [HiFiBerry DAC+ HD HiFi pcm179x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

root@b998d39a2892:/usr/src# dmesg | grep hifi
[    5.609991] snd-rpi-hifiberry-dacplushd soc:sound: ASoC: failed to init link HiFiBerry DAC+ HD: -517
[    5.635920] snd-rpi-hifiberry-dacplushd soc:sound: pcm179x-hifi <-> fe203000.i2s mapping ok
[    5.635941] snd-rpi-hifiberry-dacplushd soc:sound: ASoC: no DMI vendor name!

root@b998d39a2892:/usr/src# dmesg | grep snd
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:86:C4:5B vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=null rootfstype=ext4 rootwait vt.global_cursor_default=0 root=UUID=ba1eadef-5494-4774-a661-ffef7db248e8 rootwait
[    5.609991] snd-rpi-hifiberry-dacplushd soc:sound: ASoC: failed to init link HiFiBerry DAC+ HD: -517
[    5.635920] snd-rpi-hifiberry-dacplushd soc:sound: pcm179x-hifi <-> fe203000.i2s mapping ok
[    5.635941] snd-rpi-hifiberry-dacplushd soc:sound: ASoC: no DMI vendor name!

root@b998d39a2892:/usr/src# pactl list sinks
Sink #0
        State: SUSPENDED
        Name: alsa_output.0.stereo-fallback
        Description: Built-in Audio Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 6
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
[... 78 lines cut ...]
                pcm

Sink #3
        State: SUSPENDED
        Name: snapcast
        Description: Null Output
        Driver: module-null-sink.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 24
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: snapcast.monitor
        Latency: 0 usec, configured 0 usec
        Flags: DECIBEL_VOLUME LATENCY 
        Properties:
                device.description = "Null Output"
                device.class = "abstract"
                device.icon_name = "audio-card"
        Formats:
                pcm

Sorry mate

shawaj, 2020-12-29:
Could you post the logs from the audio container?

You can get them easily by clicking the little logs button next to the audio container like below image:
[image]

shawaj, 2020-12-29:
@arjan5340... I've spotted the issue I think.

The udev rules don't currently have an entry for `snd-rpi-hifiberry-dacplushd`
https://github.com/balenablocks/audio/blob/master/udev/95-balena-audio.rules

Relates to #296 

Will submit a PR to fix this later.

shawaj, 2020-12-29:
PR is here https://github.com/balenablocks/audio/pull/48

arjan5340, 2020-12-29:
Dear Aaron,
what is the best way for me to use your update?

arjan5340, 2020-12-29:
29.12.20 13:01:29 (+0100)  audio  Setting audio routing rules. Note that this can be changed after startup.
29.12.20 13:01:29 (+0100)  audio  Routing 'balena-sound.input' to 'snapcast'.
29.12.20 13:01:29 (+0100)  audio  Routing 'balena-sound.output' to 'alsa_output.dac.stereo-fallback'.
29.12.20 13:01:29 (+0100)  audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
29.12.20 13:01:29 (+0100)  audio  E: [pulseaudio] module-loopback.c: No such sink.
29.12.20 13:01:29 (+0100)  audio  E: [pulseaudio] module.c: Failed to load module "module-loopback" (argument: "latency_msec=200 source="balena-sound.output.monitor" sink="alsa_output.dac.stereo-fallback""): initialization failed.
29.12.20 13:01:29 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
29.12.20 13:01:29 (+0100)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
29.12.20 13:01:29 (+0100)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
29.12.20 13:15:04 (+0100)  audio  
29.12.20 13:15:04 (+0100)  audio  --- Audio ---
29.12.20 13:15:04 (+0100)  audio  Starting audio service with settings:
29.12.20 13:15:04 (+0100)  audio  - Pulse log level: NOTICE
29.12.20 13:15:04 (+0100)  audio  - Default output: AUTO
29.12.20 13:15:04 (+0100)  audio  
29.12.20 13:15:04 (+0100)  audio  Detected audio cards:
29.12.20 13:15:04 (+0100)  audio  0 bcm2835-jack bcm2835_headphonbcm2835Headphones-bcm2835Headphones
29.12.20 13:15:04 (+0100)  audio  
29.12.20 13:15:04 (+0100)  audio  
29.12.20 13:15:15 (+0100)  audio  Setting audio routing rules. Note that this can be changed after startup.
29.12.20 13:15:15 (+0100)  audio  Routing 'balena-sound.input' to 'snapcast'.
29.12.20 13:15:15 (+0100)  audio  Routing 'balena-sound.output' to 'alsa_output.bcm2835-jack.stereo-fallback'.
29.12.20 13:15:15 (+0100)  audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
29.12.20 13:15:17 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
29.12.20 13:15:17 (+0100)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
29.12.20 13:15:17 (+0100)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
29.12.20 13:17:04 (+0100)  audio  
29.12.20 13:17:04 (+0100)  audio  --- Audio ---
29.12.20 13:17:04 (+0100)  audio  Starting audio service with settings:
29.12.20 13:17:04 (+0100)  audio  - Pulse log level: NOTICE
29.12.20 13:17:04 (+0100)  audio  - Default output: AUTO
29.12.20 13:17:04 (+0100)  audio  
29.12.20 13:17:04 (+0100)  audio  Detected audio cards:
29.12.20 13:17:05 (+0100)  audio  0 sndrpihifiberry HifiberryDacplu-snd_rpi_hifiberry_dacplushd
29.12.20 13:17:05 (+0100)  audio  
29.12.20 13:17:05 (+0100)  audio  
29.12.20 13:17:15 (+0100)  audio  Setting audio routing rules. Note that this can be changed after startup.
29.12.20 13:17:15 (+0100)  audio  Routing 'balena-sound.input' to 'snapcast'.
29.12.20 13:17:15 (+0100)  audio  Routing 'balena-sound.output' to 'alsa_output.dac.stereo-fallback'.
29.12.20 13:17:16 (+0100)  audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
29.12.20 13:17:17 (+0100)  audio  E: [pulseaudio] module-loopback.c: No such sink.
29.12.20 13:17:17 (+0100)  audio  E: [pulseaudio] module.c: Failed to load module "module-loopback" (argument: "latency_msec=200 source="balena-sound.output.monitor" sink="alsa_output.dac.stereo-fallback""): initialization failed.
29.12.20 13:17:17 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
29.12.20 13:17:17 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.bcm2835-jack.stereo-fallback does not exist.
29.12.20 13:17:17 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
29.12.20 13:17:17 (+0100)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
29.12.20 13:17:17 (+0100)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
29.12.20 13:26:19 (+0100)  audio  
29.12.20 13:26:19 (+0100)  audio  --- Audio ---
29.12.20 13:26:19 (+0100)  audio  Starting audio service with settings:
29.12.20 13:26:19 (+0100)  audio  - Pulse log level: NOTICE
29.12.20 13:26:19 (+0100)  audio  - Default output: AUTO
29.12.20 13:26:19 (+0100)  audio  
29.12.20 13:26:19 (+0100)  audio  Detected audio cards:
29.12.20 13:26:19 (+0100)  audio  0 sndrpihifiberry HifiberryDacplu-snd_rpi_hifiberry_dacplushd
29.12.20 13:26:19 (+0100)  audio  
29.12.20 13:26:19 (+0100)  audio  
29.12.20 13:26:30 (+0100)  audio  Setting audio routing rules. Note that this can be changed after startup.
29.12.20 13:26:30 (+0100)  audio  Routing 'balena-sound.input' to 'snapcast'.
29.12.20 13:26:30 (+0100)  audio  Routing 'balena-sound.output' to 'alsa_output.dac.stereo-fallback'.
[... cut ...]
29.12.20 13:26:32 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
29.12.20 13:26:32 (+0100)  audio  E: [pulseaudio] main.c: Sink alsa_output.dac.stereo-fallback does not exist.
29.12.20 13:26:32 (+0100)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
29.12.20 13:26:32 (+0100)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11

shawaj, 2020-12-29:
@arjan5340 

Go to the audio container terminal and enter:
```
nano /etc/udev/rules.d/95-balena-audio.rules
```

Then add the following line:
```
DRIVERS=="snd-rpi-hifiberry-dacplushd", ATTR{id}="dac"
```
Directly after this line:
```
DRIVERS=="cygnus-ssp", ATTR{id}="dac"
```

And add the following line:
```
DRIVERS=="snd-rpi-hifiberry-dacplushd", ENV{PULSE_NAME}="dac"
```
Directly after this line:
```
DRIVERS=="cygnus-ssp", ENV{PULSE_NAME}="dac"
```

Then press control-X, then y, then enter to save and exit.

Then try restarting the audio container.

I think this should get it working, however I am unable to test as I do not have one of these sound cards here to test with.

arjan5340, 2020-12-29:
IT WORKS YES YES YES YES 
You are great It works well done mate really really well 
thanks

shawaj, 2020-12-30:
@arjan5340 awesome! I'm very glad to hear that. I was starting to get a bit annoyed that we couldn't get it to work haha.

Thanks for all your help debugging. Makes it harder not having the device here :-)

Enjoy your HiFiBerry and balenaSound setup!

I'll leave this issue open for now until the PR is merged on the audio repo.

arjan5340, 2020-12-30:
Dear Aaron,

Again thanks a lot. 2 of the 3 rasps work now i setup now the last one. 
Have a good new year en healthy 2021 for you and your family.

arjan5340, 2020-12-30:
All devices works perfectly. The sync was in the begin not very well after a couple of restarts works everything perfect.

shawaj, 2020-12-30:
@arjan5340 glad to hear you are up and running and everything working nicely.

Hope you have a great 2021 as well!

arjan5340, 2020-12-31:
Dear Aaron,

After your update i have a little problem with multiroom playing. I have 3 raspberry but the multiroom doesn't work any more this morning. I have reboot the 3 devices without any succes. I delete the devices and install them again from scratch and use your patch also. But the multiroom function doesn't work. Do you have any idea's
Best regards
Arjan

arjan5340, 2020-12-31:
logarithmic
31.12.20 12:09:57 (+0100)  spotify          --autoplay      autoplay similar songs when your music ends.
31.12.20 12:09:57 (+0100)  spotify          --disable-gapless 
31.12.20 12:09:57 (+0100)  spotify                          disable gapless playback.
31.12.20 12:09:57 (+0100)  spotify  
31.12.20 12:09:57 (+0100) Service exited 'spotify sha256:06b3597304ff8744f0018939fdd6a166ea64baba43b4fd0646ba941f7c933536'

arjan5340, 2020-12-31:
Now i can't see the devices on spotify any more

arjan5340, 2020-12-31:
31.12.20 13:12:49 (+0100) Failed to download image 'registry2.balena-cloud.com/v2/5bb6966706680ddac2361d7d57331a98@sha256:0d7c592579307e9e4dab42b9403407aad9ac50bcbb39f8039ffc6435a3a20b2a' due to 'failed to register layer: Error processing tar file(exit status 1): write /lib/udev/rules.d/60-block.rules: no space left on device'

I think above is the problem how can i clean my hole rasp? and start again

richardpringle, 2020-12-31:
Don't the file edits disappear after restarting the container? I added the lines, restarted the container, everything appears to be working, but when I went to checkout the same file, my edits were gone. 

🤷

shawaj, 2021-01-01:
@richardpringle yes unfortunately that is the case until the changes are pulled in.

@arjan5340 your first issue seems to be related to #387 / #386 

The second one seems to be that your SD card is full. Perhaps you have a larger sd card or can wipe that one and start again?

aandroide, 2021-01-01:
> This might be interesting to try also [#357 (comment)](https://github.com/balenalabs/balena-sound/issues/357#issuecomment-751471439)
> 
> However, as I said I don't have a board to test it.

it has been determined that there are errors in the 91 rules file. by pulseaudio rpi2.
this caused the assignment of audio cards to go haywire.
the 2 solutions are either delete the file with sudo rm /lib/udev/rules.d/91-pulseaudio-rpi.rules
or replace the content with:

SUBSYSTEM!="sound*", GOTO="end"
--
  | ACTION!="change", GOTO="end"
  | KERNEL!="card*", GOTO="end"
  | ENV{SOUND_FORM_FACTOR}!="internal", GOTO="end"
  |  
  | DRIVERS=="bcm2835_audio", ATTR{id}=="b1", ENV{PULSE_PROFILE_SET}="rpi-hdmi.conf", GOTO="end"
  | DRIVERS=="bcm2835_audio", ATTR{id}=="b2", ENV{PULSE_PROFILE_SET}="rpi-hdmi.conf", GOTO="end"
  | DRIVERS=="bcm2835_audio", ATTR{id}=="Headphones", ENV{PULSE_PROFILE_SET}="rpi-analog.conf", GOTO="end"
  |  
  | LABEL="end"

``
this should solve the problem

shawaj, 2021-01-01:
@aandroide where are you seeing these issues? Is there a PR somewhere you can refer to?

aandroide, 2021-01-01:
> @aandroide where are you seeing these issues? Is there a PR somewhere you can refer to?
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=294600&p=1789619#p1789619

aandroide, 2021-01-01:
> @aandroide where are you seeing these issues? Is there a PR somewhere you can refer to?

https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1082#note_739409

shawaj, 2021-01-01:
@aandroide how does this relate to the hifiberry dac hd2 not working? I can't see how it relates? It seems to be a different issue?

aandroide, 2021-01-01:
> @aandroide how does this relate to the hifiberry dac hd2 not working? I can't see how it relates? It seems to be a different issue?

I'm just reporting that whoever has the image of december buster and therefore pulseaudio updated will surely have problems due to an error in the 91 rules file. in my case I had a google voice hat and I solved the problem by making the changes declared before. mine is just a suggestion and not a problem here.

shawaj, 2021-01-01:
@aandroide ahh I see - ok thanks for the heads up. We don't use Raspberry Pi OS because this runs on balenaOS which is yocto based. So not sure we will be affected by this.

We don't have a file at `/lib/udev/rules.d/91-pulseaudio-rpi.rules`:
[image]

shawaj, 2021-01-01:
@arjan5340 @richardpringle the audio block repo is now updated with the latest images.

Could you try rebuilding and pushing?

https://hub.docker.com/r/balenablocks/audio/tags?page=1&ordering=last_updated

shawaj, 2021-01-01:
Seems to be working for me:

[image]

shawaj, 2021-01-01:
@arjan5340 has confirmed in this thread that it is now working https://github.com/balenablocks/audio/pull/48

So closing this issue now.

closed 2021-01-01 by shawaj