iotsound/iotsound #378: Belena Sound

alex4buba, 2020-12-23:
Can this be a replacement as a SONOS music controller for KDE Plasma?
If so, how?

Thanks
Alex

shawaj, 2020-12-25:
@alex4buba do you mean that you have Sonos devices you want to control from your desktop / laptop computer?

If you could provide a bit more detail about what you're trying to achieve that would be great.

alex4buba, 2020-12-25:
Hello Aaron
Well, I have the Sonos system installed with 6 devices for the past 5 years. I
started it then with a Windows base PC and the company provides the software
controller for it. Some 8 months ago, I had enough of Windows and switched to
Linux. I tried first Ubuntu 20.05 LTS, went through several other distros and
settled on using now KDE Neon Plasma 5.20.4
I am using an app called NOSON, it kind of works....
Not sure what other information  I need to provide.
I am looking for an app to work on my current Linux / KDE desktop - to be able
to control those Sonos units
CheersAlex

shawaj, 2020-12-25:
@alex4buba you could maybe try looking at https://roonlabs.com/

Basically balenaSound is a replacement for Sonos using speakers that are not internet connected directly. It allows you to use a Raspberry Pi to upgrade your "dumb" speakers to bring "smart"

Having said that, you could set this up to bypass / work concurrently with your Sonos system by adding a Raspberry Pi to the aux input of each Sonos speaker that you have.

Hope that makes sense.

alex4buba, 2020-12-25:
Hello again,

No, not clear. I have no idea what is a Raspberry pi?

Thanks again
Alex

shawaj, 2020-12-25:
@alex4buba a Raspberry Pi is a computer that runs a version of Linux. balenaSound is designed to work with Raspberry Pi.

You can read more about how balenaSound works and what it is used for here https://sound.balenalabs.io/docs

This would bypass the Sonos control system and allow balenaSound to output music directly to your Sonos speakers via the aux inputs on each speaker.

AlexProgrammerDE, 2020-12-29:
Balenasound is intended as a complete replacement for sonos. It also has multi-room capabilities. Im gonna close this issue for now.

closed 2020-12-29 by AlexProgrammerDE