iotsound/iotsound #379: AirPlay crashing after some time

SteffenDE, 2020-12-24:
I've noticed AirPlay crashing after a while in idle multiple times in the last few days. I'm seeing the following error in the balena Logs:
```
fatal error: could not connect to the pulseaudio playback stream -- the error message is "Bad state".
```
Restarting the airplay container fixes this problem.

**Device information**
 - Board (hardware info): Raspberry Pi 4b
 - Device type (as seen in balenaCloud): Raspberry Pi 4
 - balenaOS version: balenaOS 2.65.0+rev1

**Additional context**
The device is operating in standalone mode and has the following variables set:
```
SOUND_DISABLE_SPOTIFY=1
SOUND_MODE=STANDALONE
SOUND_VOLUME=100
SOUND_DEVICE_NAME=my-custom-name
```

jacobhq, 2021-07-03:
Any progress on this one? I’m encountering the same problem with the same hardware also.

closed 2021-11-24 by bulldozer-balena[bot]

tmigone, 2021-11-24:
Fixed in v3.6.11. For details see https://github.com/balenalabs/balena-sound/pull/508