iotsound/iotsound #383: Add Infineon / Merus Audio amp board

shawaj, 2020-12-26:
The [Merus Audio Amp pHAT by Infineon] is a super efficient amplifier for Raspberry Pi (can output 40W peak using just the 2.5A Pi power supply!). It uses the [MA12070P](https://www.infineon.com/cms/en/product/power/class-d-audio-amplifier-ic/integrated-class-d-audio-amplifier-ics/ma12070p/) chip and the `merus-amp.dtbo` device tree overlay.

Working on a PR - but just adding this to track progress/info.

```
root@2ac79ccab9fe:/usr/src# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: dac [snd_rpi_merus_amp], device 0: Merus Audio Amp ma120x0p-amp-0 [Merus Audio Amp ma120x0p-amp-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

```
root@2ac79ccab9fe:/usr/src# cat /proc/asound/cards
 0 [dac            ]: RPi-simple - snd_rpi_merus_amp
                      snd_rpi_merus_amp
```

```
root@2ac79ccab9fe:/usr/src# cat /proc/asound/modules
 0 snd_soc_rpi_simple_soundcard
```

shawaj, 2020-12-26:
[Infineon-MA12070P-DS-v01_00-EN.pdf](https://github.com/balenalabs/balena-sound/files/5742607/Infineon-MA12070P-DS-v01_00-EN.pdf)

[Infineon-KIT_40W_AMP_HAT_ZW-UserManual-v01_00-EN.pdf](https://github.com/balenalabs/balena-sound/files/5742608/Infineon-KIT_40W_AMP_HAT_ZW-UserManual-v01_00-EN.pdf)

[Infineon-KIT_40W_AMP_HAT_ZW-ApplicationNotes-v01_00-EN.pdf](https://github.com/balenalabs/balena-sound/files/5742611/Infineon-KIT_40W_AMP_HAT_ZW-ApplicationNotes-v01_00-EN.pdf)

[Infineon / Merus GitHub page](https://github.com/Infineon/merus-audio-amp-hat-zw)

shawaj, 2020-12-26:
Was having the same issue as described [here](https://github.com/Infineon/merus-audio-amp-hat-zw/issues/4#issuecomment-724426136)

Basically - the resolution is to edit the dtparams to be:
`"i2c_arm=on","spi=off","audio=off"`

As SPI pin CE0 (pin 24 / GPIO8) is attached to MSEL0 it causes issues if the SPI interface on the Pi is on.

[image]

Also for reference see https://github.com/Infineon/merus-audio-amp-hat-zw/issues/4 and https://github.com/raspberrypi/linux/pull/4027

shawaj, 2020-12-26:
Added a PR on balena-raspberrypi repo to pull in the merus-amp device tree overlay to balenaOS - https://github.com/balena-os/balena-raspberrypi/pull/585 (as well as a few other audio related ones such as the IQaudIO codec)

tmigone, 2021-04-20:
Board now supported in balenaOS.
Superceded by https://github.com/balenalabs/balena-sound/issues/439

closed 2021-04-20 by tmigone