iotsound/iotsound #385: DAC Support

kcstauffer21, 2020-12-29:
I just purchased RPI HiFi AMP HAT TAS5713 Amplifier off of amazong (link below) I don't believe it is compatible with balena. Can I get the support added? 

Amazon link: https://www.amazon.com/gp/product/B07CZZ95B9/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Thanks,
Kendall

shawaj, 2020-12-30:
@kcstauffer21 we have only been able to test DACs where the team already had them available or where the manufacturer have sent us them for testing. The list we have isn't a list of all DACs that work with balenaSound - it's a list of tested DACs that we know work.

In any case, looking at [the documentation for your Amp board](http://wiki.inno-maker.com/plugins/servlet/mobile?contentId=1147008#content/view/6948250) it appears that we already support it.

You just need to add the dtoverlay `hifiberry-amp` in the "Define DT overlays" section under the device configuration menu section.

Please let us know if this works and we can get the board added to the list for future users :-)

kcstauffer21, 2020-12-31:
It works thanks!

shawaj, 2020-12-31:
@kcstauffer21 great will add to the supported DACs list shortly.

Just for future reference the official name is the Innomaker HiFi Amp HAT

closed 2021-01-01 by bulldozer-balena[bot]

cheakheang, 2021-04-18:
i am new to Balena, i have same board  "HiFi AMP HAT TAS5713 DAC" as above, i try add dtoverlays as below, but there is no sound, please help me on this issue. thank you.

[image]

PR #389 (2021-01-01): docs: add various dacs that have been tested
adds instructions for various dacs that have been tested to work with balenaSound

Closes: #385 
Closes: #355 
Closes: #345 
Change-type: patch
commit 99d2e787 (2021-01-01): docs: add various dacs that have been tested

adds instructions for various dacs that have been tested to work with balenaSound

Closes: #385 
Closes: #355 
Closes: #345 
Change-type: patch